prusa3d/PrusaSlicer #14332: Crash when text tool closed with no text

murk-sy, 2025-03-22:
### Description of the bug

When editing a text and accidentally/intentionally closing the text tool with no text written (or just a space), Prusaslicer crashes.

Ideally, blank text entries should be allowed for marking some parts and not others, or at least reset to some default text (like "Text")

### Project file & How to reproduce

[empty_text_crash.zip](https://github.com/user-attachments/files/19401526/empty_text_crash.zip)

1) Open 3MF
2) Remove all text or enter just a space
3) Click outside the text edit box

### Checklist of files included above

- [x] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.9.1-win64

### Operating system

Windows 10

### Printer model

Mini+

TerPuc, 2025-03-24:
Dear @murk-sy, thank you for your report. Internal issue SPE 2727 was created.

murk-sy, 2026-09-24:
Resolved as of 3.0.0-alpha12, it reverts to previously entered text.

That also removes any changes made if no text entered and then the menu is exited, but I think that's fine.

closed 2026-09-24 by murk-sy