prusa3d/PrusaSlicer #14333: Emboss text tool closes when ending drag-select outside the tooltip

murk-sy, 2025-03-22:
### Description of the bug

For selecting existing text and removing it, I've been using muscle memory to just drag select and then backspace to delete the text. This immediately closes the tool.

On the emboss text tool, the same behaviour is exhibited on Height, Depth, and other text-entry compatible fields.

**However, this behaviour is inconsistent with left toolbar tools and even SVG entry**, where the text-entry compatible fields can be drag selected, with the drag click being depressed outside the tool.

### Project file & How to reproduce

Click and hold left mouse key, and drag select all the text, ending outside the tool, then deselect

[image]

[image]

### Checklist of files included above

- [ ] Project file
- [x] Screenshot

### Version of PrusaSlicer

2.9.1-win64

### Operating system

Windows 10

### Printer model

Mini+

TerPuc, 2025-03-27:
Dear @murk-sy, thank you for your report. Internal issue SPE 2730 was created.

murk-sy, 2026-09-24:
Resolved as of 3.0.0-alpha12

closed 2026-09-24 by murk-sy