prusa3d/PrusaSlicer #14334: Hollow feature for FDM printing (according to meshmixer's hollow feature and Prusa Slicer SLA hollow feature)

Demolux, 2025-03-22:
**Is your feature request related to a problem? Please describe.**
We do a lot of hollow sculptures. This is useful e. g. for large multi-part models as welding/glueing e. g. 5 mm shells together is more "precise" than having large flat surfaces from "blocks" touching each other. Also, you can fill it with concrete and the overall "feeling" of sturdiness is increased when using say a 8 mm shelled/walled object with say 2 perimeters and 15% infill instead of using those same perimeters and infill on the unhollowed object.

**Describe the solution you'd like**
I would like to have an option to hollow out my parts with varying degree of resolution and mesh confidence (just like in meshmixer).

**Describe how it would work**
Use and enhance the hollow feature of the SLA part of Prusa Slicer, use meshmixer's hollow tool as a guideance considering the parameters. The hole feature is not needed as you can use subtraction bodies - its already much more flexible than in meshmixer.

**Describe alternatives you've considered**
Currently I use meshmixer to generate an "inner body" by hollowing out my sculpture, deleting the outer surface and inverting the normals on the inner surface. I then modify that body to my liking (e.g. smooth sharp spikes). Then I export that and either do a boolean subtraction in 3D builder or use that part as a subtraction body in Prusa Slicer.

**Additional context**
This workflow gets especially anoying when you want different sizes of your model but with the same wall thickness. You need to repeat it for every size then. This is what I want to achieve in the end (small model example):

[image]

u89djt, 2025-03-22:
(From a fellow user) You can export the processed plate of an SLA job as an stl file.

[image]

Demolux, 2025-03-22:
Thanks! Though that would not be as convenient as it could be tbh - it would come close to my current process or any other modelling software or resin slicer.
I more thought about doing it directly in the fdm part of Prusa Slicer, best would be if its "parametric" or non-invasive, e. g. using a negative volume defined by the shell thickness. This would also circumvent the support problem - afaik there is no support generation then, in surfaces generated from negative volumes.

u89djt, 2025-03-22:
What do you actually want in terms of supports? Can you post an example where you can't achieve what you need with the slicer as it is now?

Demolux, 2025-03-22:
Sure!
Take a look here:

[image]

The left cube is hollowed out with a negative volume (see the part UI on the left). The right cube is hollowed out natively. Both cubes have been autopainted with "Paint on support". 

This is how that looks like on the left cube - no supports painted as its shell, its inner surface, is defined by a negative volume:

[image]

This is how that looks like on the right cube - supportes painted in some way (doesn't matter how) as its shell, its inner surface, defined by geometry.

[image]

Now. Lets go an generate supports. All standard settings here. There is the following behavour using supports:

1. Using "None". Obviously.
[image]

3. Using "Support on build plate only". Why does it generate not on build plate actually? Behaviour not as expected in my opinion.
[image]

3. Using "For support enforcers only". This is how its done automatically. Support inside the model where you don't want it to be. So, if your model is more complicated that only leaves manual painting as an option - or deleting the auto generated paint inside the model, but that is a pain to do on larger models.
[image]

4. Using "Everywhere". Also obviously.
[image]

So what I would like:

1. Differentiate between the inner and outer surface of a body considering automatic support generation and automatic support painting.
2. Really only generate support on build plate, also when using support enforcers - and dismiss them when they are not on the build plate. That would be kind of a workaround for 1., BUT there could be models of different geometry which still would make 1. necessary, e. g. with a very narrow opening on the bottom where support is generated "through", spreading out growing inside the upper, larger inner volume.
3. Another possible feature: Consider surfaces defined by negative volumes also for support generations.

I hope that makes it a bit more clear? I know its a bit.... complicated

u89djt, 2025-03-22:
You can paint support blockers inside a hollow object by moving the "clipping of view" slider:

[image]

Demolux, 2025-03-22:
> You can paint support blockers inside a hollow object by moving the "clipping of view" slider:

Yes I know. But that is really complicated. Sometimes you just want auto-supports to save time. There are geometries which make it VERY hard to use that clipping slider, especially when the model is larger (slider resolution gets sparse) and then its very organic and with lots of cavities.

So, I would like to see 2 options considering support generation:
1. Use negative volume surfaces for support generation
2. Exclude inner surfaces from support generation

In addition to a hollow feature. That is the main feature I am missing actually.

u89djt, 2025-03-22:
Aye, I'm focusing on solutions.
Finer control of clipping:
[image]
Post your 3mf file where it's supporting not on the build plate when you told it otherwise?

Demolux, 2025-03-22:
I guess the painted supports just overrides the support only on build plate feature.

Tbh the clipping solution itself is essential to control inner supports at all, but thats about it. I think a checkbox with the options

-     Include negative volume surfaces for support generation
-     Exclude inner surfaces from support generation

somewhere in support settings would make life easier and UX nicer in specific use cases when using auto generation or auto painting. Of course it should be locked behind advanced mode as its only needed for some advanced stuff. This way it would not clutter the menue.

Also, the solution for finer clipping control isn't a very nice UX. Holding shift for finer control of the slider would be, but not entering a text field :/

And aside from that, the actual feature I prefer over all the support generation UX refinements is **hollowing in FDM mode** :)

u89djt, 2025-03-22:
Modifier keys for drag targets make a lot of sense. 
If you post the 3mf file we'll be able to work it out. There may be a function you're not aware of yet.

Demolux, 2025-03-23:
I can't upload .3mf's here but here is a DL link to that simple cube file. 
https://limewire.com/d/o4US7#dDBL0K0tAC

Be aware that this is only a simplified example. The slider works well with it as it is small and you only have one flat inner face where support generates. This can look a lot different, see here:

[image]

That is an example which is large and it has a small channel connecting its inner surface with the outer one. Tree support generates through it and spreads inside. Catching all the auto painted surfaces and deleting them inside was a real pain. But it was less pain than painting the surfaces from outside completly manually. I cannot post that file though.

u89djt, 2025-03-23:
If you want to post a 3mf file - obviously not something with delicate IP - you can zip the file. If you want a developer to take an interest, you need to demonstrate potential value and place explanatory files here.
So you're not actually talking about an inner surface. You or the slicer would have to decide what constitutes an inner cavity that happens to have one or more holes somewhere.
You can plug gaps with 0.2mm high blocks to be poked out, Add a cylinder as a part to the object and you can scale it and move it in 3D.

Demolux, 2025-03-23:
I actually primarily wish for the hollow feature. I should have posted the support enhancement seperatly as this topic became heavy on it. I guess it's clear how a hollow feature could work as its already present in Prusa Slicer but locked for FDM slicers.

So maybe I will make a dedicated post considering the support thing - actually two, one for each of the two options. Let's focus on the hollow feature, and in my understanding there is nothing really left to explain is it?

Tupson444, 2026-09-19:
still relevant