prusa3d/PrusaSlicer #14337: SVG on object

SpudBoxTeam, 2025-03-22:
### Description of the bug

when I add an SVG icon to my model when I scale it down to 1 on the depth then make the SVG smaller and put it into use surface it crashes MacBook Air 2021 m1 8gb ram 

### Project file & How to reproduce

go to prusa slicer 

import a largish object 

insert an SVG onto the object 

make depth 1mm 

make the object smaller 

then click use surface 

it will crash 

issue occurred Mac OS

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

Version 2.9.1

### Operating system

MacOS 15.3.2

### Printer model

Prusa MK4S HF 0.4mm nozzle

u89djt, 2025-03-22:
(From a fellow user) Have a look at the bug report guidance. You'll have to zip the 3mf and svg files to post them.

SpudBoxTeam, 2025-03-26:
It won't give me a bug report Prusa slicer just closes

SpudBoxTeam, 2025-03-26:
Thanks I will add a model and SVG in a zip file later

JasonArh, 2025-03-26:
I'm having the same issue. It's constantly crashing whenever I try to emboss with and SVG.
It been an issue for the past 3 versions or so. Even happens when just doing text.
It crashes when clicking use surface. Or if it happened to not crash when use surface has been checked it will crash if I try to move the svg or text or also if I resize it.
2.9.1 on Mac.

u89djt, 2025-03-26:
@JasonArh they won't be able to help until you post a 3mf file of the situation just before you click "use surface" so they have your context. They obviously haven't had the same problem, so you need to put them in your shoes. You need to zip 3mf files to post them.

JasonArh, 2025-03-26:
It's really not hard to replicate. Here's a file with a generic cylinder embossing some text on it. Just have to adjust the text size while "use surface" is check and it crashes. 

[Text Crash.3mf.zip](https://github.com/user-attachments/files/19474665/Text.Crash.3mf.zip)

JasonArh, 2025-03-26:
Here's another file using an SVG. Generic cylinder embossing an SVG. It'll wrap fine but if you adjust size after "Use Surface" is checked it will crash. 

I've had other files crash when selecting "Use Surface" Other times while moving the modifier around. Some times it will crash if you move it and you have it set as a cut. Sometimes it will crash when you just change it from a modifier to "Cut".

[SVG Crash.3mf.zip](https://github.com/user-attachments/files/19474801/SVG.Crash.3mf.zip)

u89djt, 2025-03-26:
The idea is to not make them spend time replicating the issue, and instead put them directly in the right place and context with the files you provide. You're reporting a bug that they haven't seen, so it doesn't happen to them in the normal course of operations. The bug submission guidance is provided for folk not accustomed to bug hunting.

closed 2026-09-29 by prusaslicer-bot[bot]