prusa3d/PrusaSlicer #14342: Consistent outer wall detection

SXRguyinMA, 2025-03-23:
I'd like to have a setting that would make the travel direction for outer (or inner for holes) the same for every layer.

It would work buy having a check box or drop down to enable/disable the setting.

The issue is when printing with coextruded filaments (Matterhackers Quantum for example) the colors can mix when you don't want them to depending on what direction the extruder is traveling resulting in uneven colored surfaces.

With a round part for example one layer will have the outer perimeter go clockwise and other times it'll go counter-clockwise. This feature would force it to always go CW or CCW for the outer wall (or inner wall for a hole or other inside hole-style feature).

neophyl, 2025-03-23:
Duplicate of https://github.com/prusa3d/PrusaSlicer/issues/7575  or https://github.com/prusa3d/PrusaSlicer/issues/10044

Jan-Soustruznik, 2025-04-10:
The reported feature request is similar to #7575. Collected feature request. Closed as duplicate.

closed 2025-04-10 by Jan-Soustruznik