prusa3d/PrusaSlicer #14346: Multiple problems with dovetail cut

mi3dke, 2025-03-23:
### Description of the bug

I want to print an object in different parts using cut with the dovetail option.
There is a base and three rips that I want to cut off. 

[image]

First problem is that after cutting I cannot split the three rips into individual objects to position them individually with their backs on the bed.

[image]

If I perform the cut creating only Object A I can perform a split.

[image]

But when the split is completed the dovetails are also individual objects and moreover their positions are changed (re-nulled?).

[image]

### Project file & How to reproduce

[Chinmount GoPro Adapter_Report.zip](https://github.com/user-attachments/files/19411319/Chinmount.GoPro.Adapter_Report.zip)
Select the object.
Perform cut using these settings.

[image]

**Problem A: Object A cannot be split.**

Undo and perform cut again, this time delecting Object B.

Split Object A into objects. 
**Problem B: dovetails are treated as individual Objects.
Problem C: Dovetails positions change.** 

### Checklist of files included above

- [x] Project file
- [x] Screenshot

### Version of PrusaSlicer

2.9.1 windows

### Operating system

Windows 10

### Printer model

Prusa MK3S+

neophyl, 2025-03-24:
You just cant use split.  Creating the dovetail creates multiple parts.  When you split then of course it moves them.  However if you approach it differently you can do what you want.   I had no trouble doing what you want just using successive cuts.

[image]

u89djt, 2025-03-24:
(From a fellow user) This happens because the slicer retains a view of the whole assembly so that you can modify the cut. You need to tell the slicer to forget about it by choosing "invalidate cut information"
[image]
[image]
To get everything lying on its back takes a bit of work. First, select the entity you want to break down and rotate it so your individual components are in their final orientation.
[image]
Then split to PARTS. Then select pairs of item and dovetail and move them out of the way.
[image]
[image]
[image]
Then split to OBJECTS and they drop to the plate.
[image]
Then group your items with the corresponding dovetail.
[image]
You can now position them as necessary.
[image]

neophyl, 2025-03-24:
Much easier to just cut the tabs with 2 more planar cuts after the dovetail cut is made and then use place to face on them.  No need to use split to either parts or objects.

u89djt, 2025-03-24:
What did you mean by "differently" and "successive cuts" earlier?

neophyl, 2025-03-24:
Differently and successive means exactly what you just did.  Differently- not using 'split to' at all.  Using a different process to achieve the desired goal.  And 'successive cuts' means to apply multiple cuts one after another to chop the top tabs up into distinct 'pieces'.  Trying tom avoid the term Object or Part as those already have very specific meaning as far as PS is concerned.  

Language can be so messy.

u89djt, 2025-03-24:
Ah, I read "tabs" as the dovetail tabs.

mi3dke, 2025-03-24:
Thanks for your suggestions.
It was less about finding alternative methods rather than understand, why split to objects is not working or why "it has to be" that the dovetails are ending up being an idependent object as well.
Nonetheless interesting to see how you solved it. One can always learn watching others.

I solved it by copying and moving the ribs in z-axis and used negative volumes. 

[image]

But in the end all solutions are not what I would wish for, because once you want to change a detail on the dovetail-cut, you start repeating all steps from the beginning.

closed 2026-09-29 by prusaslicer-bot[bot]