prusa3d/PrusaSlicer #14353: single-instance-on-url

yahbluez, 2025-03-24:
### Description of the bug

The linux flatpak version 2.9.1 of prusaslicer 
ignores the 

<Configuration><Preferences><Allow just a single PrusaSlicer Instance>unchecked

setting and has a hardcoded:

Exec=entrypoint --single-instance-on-url %u

in his flatpak desktop setting

[image]

### Project file & How to reproduce

No Project or File needed. 

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

flatpak 2.9.1

### Operating system

linux

### Printer model

core one

SachCZ, 2025-03-24:
Hello, I do not understand what are you referring to as unchecked.

yahbluez, 2025-03-24:
This: 

[image]

If no PrusaSlicer instance is running URL download starts one.

but

if there is already an instance running, the URL download will add the content into that instance.

This is a bug.

It happens only with URL download not with filesystem based actions.
It is new older version don't have that bug.

I traced it down to the flatpak setup.
To fix this bug the exec setting needs tobe fixed in  the .desktop file.

It is line 6 in my screenshot.

SachCZ, 2025-03-25:
I still do not understand you. The exec line in the desktop file is intentional.

This

> 
> If no PrusaSlicer instance is running URL download starts one.
> 
> but
> 
> if there is already an instance running, the URL download will add the content into that instance.

exactly describes the intended functionality. What is the problem, what is different from any previous version?

yahbluez, 2025-03-25:
> What is the problem, what is different from any previous version?

If I open 2 models by url they are meshed together into one slicer instance,
instead of running each in her own instance.

Doing the same, open 2 models from the desktop file selector / explorer,
each get his own PrusaSlicer instance.

It makes completely no sense to mesh two models into one slicer instance.
Why ignores the URL opener the settings from preferences?
Just not ignoring the settings and the problem is gone.

There is no need to handle URL openings in any way different from any other file openings. 

The problem is, that if you have a running PrusaSlicer and click in your browser on some other model it will be added to the running instance.

SachCZ, 2025-03-25:
Oh, I see what you mean now. Is not a bug. It is intended, because people use printables within slicer, where it makes sense to open the model in the current instance. Also login to slicer is currently handled this way as well.

I will flag this as a feature request. To summarize, it is requested that:

- When a model is opened from internet browser it would make sense to open it in a new slicer instance.

yahbluez, 2025-03-26:
Yah, that is the point. 
Workaround is to download and than open from file explorer.

closed 2026-09-29 by prusaslicer-bot[bot]