prusa3d/PrusaSlicer #14354: Sequential Printing CRASH because of incorrect tool chnager model for Prusa XL.

Lungenpest, 2025-03-24:
### Description of the bug

I tried to print sequential objects.
The the lower clamping lever of the tool changer crashed in the previous object.

That happens because of a incorrect 3d model in Prusa Slicer 2.9.1.
The the lower clamping lever is not in the final clamping position.

The screenshot from should tell you everything you need to know.

Please fix the model.

[image]

[image]

### Project file & How to reproduce

Slice model and go to positions shown in the screenshot.

[Gridfinity-Box-1x1-VaseMode-alle5Höhen-SEQ-CRASH-ps291.zip](https://github.com/user-attachments/files/19437192/Gridfinity-Box-1x1-VaseMode-alle5Hohen-SEQ-CRASH-ps291.zip)

### Checklist of files included above

- [x] Project file
- [x] Screenshot

### Version of PrusaSlicer

2.9.1

### Operating system

Manjaro Flatpak

### Printer model

Prusa XL 5T

Jan-Soustruznik, 2025-03-27:
Internal Issue created(SPE-2731)

tneff03, 2025-03-28:
The XL model is also missing the belts. Found this out the hard way on a long sequential print today. 
The placement was done with auto arrange, then I double checked the preview and it had plenty of room. But that’s because the model needs the belts. 

You can see in the images where the tall piece gets hit by the belt. 

[image]
[image]

Lungenpest, 2025-03-28:
> This is in no way even slightly official, and has only been tested very briefly. I moved the bar in the stl and put it in the slicer's gantry models folder. It seems to work when arranging OP's problem project. [prusa3d_xl_gantry.zip](https://github.com/user-attachments/files/19486056/prusa3d_xl_gantry.zip) [image]

Could you please add the belts, too? Like @tneff03 explained? That would be great an help us so much.

u89djt, 2025-03-28:
Bear in mind I'm a fellow user - not associated with the prusa folk. I'll happily do it if I can get my hands on an accurate model to work from. The ones I've seen all look a bit approximate.

u89djt, 2025-03-28:
This is a quick hack with zero promises. I haven't tested it. 

[XL gantry hacked with belt.zip](https://github.com/user-attachments/files/19512081/XL.gantry.hacked.with.belt.zip)

Lungenpest, 2025-03-29:
> This is a quick hack with zero promises. I haven't tested it.
> 
> [XL gantry hacked with belt.zip](https://github.com/user-attachments/files/19512081/XL.gantry.hacked.with.belt.zip)

First, thanks for your volunteer work!

I replaced the gantry model in the resources folder and the model itself doesn't change the arrangement behaviour, there is again collision on the first try with the belt. didn't checked the collision on the lever again, but I'm afraid that there is collision, too. Need further inspection.

[image]

3MF Project:
[ps291-seq-crash.zip](https://github.com/user-attachments/files/19516856/ps291-seq-crash.zip)

u89djt, 2025-03-29:
Ah well, that would be why it's an internal issue rather than an immediate push from prusa. I imagine they have a simpler mesh deeper in the system.

Lungenpest, 2025-03-29:
> Ah well, that would be why it's an internal issue rather than an immediate push from prusa. I imagine they have a simpler mesh deeper in the system.
Yes, i think its in the geometrie.json file in the same directory

`        {
            "printer_notes_regex": ".*PRINTER_MODEL_XL.*",
            "gantry_model_filename": "prusa3d_xl_gantry.stl",
            "slices": [
                {
                    "height": "0",
                    "type": "convex",
                    "polygons": [
                        "-5,-5;5,-5;5,5;-5,5"
                    ]
                },
                {
                    "height": "2",
                    "type": "convex",
                    "polygons": [
                        "-10,-47;34,-47;34,16;-10,16",
                        "-34,13;32,13;32,67;-34,67"
                    ]
                },
                {
                    "height": "23",
                    "type": "convex",
                    "polygons": [
                        "-42,11;32,11;32,66;-42,66",
                        "-33,-37;43,-37;43,18;-33,18",
                        "-13,-68;47,-68;47,-30;-13,-30"
                    ]
                },
                {
                    "height": "19",
                    "type": "box",
                    "polygons": [
                        "-400,24;400,24;400,50;-400,50"
                    ]
                },
                {
                    "height": "220",
                    "type": "box",
                    "polygons": [
                        "-400,-400;400,-400;400,10;-400,10"
                    ]
                },
                {
                    "height": "280",
                    "type": "box",
                    "polygons": [
                        "-400,-400;400,-400;400,400;-400,400"
                    ]
                }
            ]
        },`

u89djt, 2025-03-29:
Try this at your own risk. I altered the block above 2mm to include the bar, and the block at 19 to include the belt.
        {
            "printer_notes_regex": ".*PRINTER_MODEL_XL.*",
            "gantry_model_filename": "prusa3d_xl_gantry.stl",
            "slices": [
                {
                    "height": "0",
                    "type": "convex",
                    "polygons": [
                        "-5,-5;5,-5;5,5;-5,5"
                    ]
                },
                {
                    "height": "2",
                    "type": "convex",
                    "polygons": [
                        "-10,-47;34,-47;34,16;-10,16",
                        "-40,13;32,13;32,67;-40,67"
                    ]
                },
                {
                    "height": "23",
                    "type": "convex",
                    "polygons": [
                        "-42,11;32,11;32,66;-42,66",
                        "-33,-37;43,-37;43,18;-33,18",
                        "-13,-68;47,-68;47,-30;-13,-30"
                    ]
                },
                {
                    "height": "19",
                    "type": "box",
                    "polygons": [
                        "-400,24;400,24;400,60;-400,60"
                    ]
                },
                {
                    "height": "220",
                    "type": "box",
                    "polygons": [
                        "-400,-400;400,-400;400,10;-400,10"
                    ]
                },
                {
                    "height": "280",
                    "type": "box",
                    "polygons": [
                        "-400,-400;400,-400;400,400;-400,400"
                    ]
                }
            ]
        },

u89djt, 2025-04-06:
Cool, and the json changed to match I guess? I made the changes below.
        {
            "printer_notes_regex": ".*PRINTER_MODEL_XL.*",
            "gantry_model_filename": "prusa3d_xl_gantry.stl",
            "slices": [
                {
                    "height": "0",
                    "type": "convex",
                    "polygons": [
                        "-5,-5;5,-5;5,5;-5,5"
                    ]
                },
                {
                    "height": "2",
                    "type": "convex",
                    "polygons": [
                        "-10,-47;34,-47;34,16;-10,16",
                        "-40,13;32,13;32,67;-40,67"
                    ]
                },
                {
                    "height": "23",
                    "type": "convex",
                    "polygons": [
                        "-42,11;32,11;32,66;-42,66",
                        "-33,-37;43,-37;43,18;-33,18",
                        "-13,-68;47,-68;47,-30;-13,-30"
                    ]
                },
                {
                    "height": "19",
                    "type": "box",
                    "polygons": [
                        "-400,24;400,24;400,60;-400,60"
                    ]
                },
                {
                    "height": "220",
                    "type": "box",
                    "polygons": [
                        "-400,-400;400,-400;400,10;-400,10"
                    ]
                },
                {
                    "height": "280",
                    "type": "box",
                    "polygons": [
                        "-400,-400;400,-400;400,400;-400,400"
                    ]
                }
            ]
        },

Lungenpest, 2025-05-16:
Tried several sequential prints since 2.9.2 . No more crashes.

closed 2025-05-16 by Lungenpest