prusa3d/PrusaSlicer #14356: Cannot Login to Prusa Account in PrusaSlicer 2.9.1 - Ubuntu Linux 24.04.2 LTS

carltondhouston, 2025-03-24:
### Description of the bug

Problem: PrusaSlicer starts and runs but has a problem logging into my Prusa account.  I hit login, enter my credentials and it immediately returns a red filament roll and a message "Something went wrong."  It doesn't matter if the credentials entered are right or wrong.

I expected it to connect to my Prusa account and permit me to send prints to my printers.

### Project file & How to reproduce

I tried creating a project file but it appears 3mf files are not allowed.  There is no print object or anything other than a bare configuration because I cannot get the connectivity to the printers to work via Prusa Connect.  So I am not sure how this might hurt my case getting attention.

Open PrusaSlicer -> Login -> Login
Enter username email address
Enter password 
Click LOG IN

Note that clicking any other buttons including "Create account" or "Forgot Password" results in the same immediate error message.

[image]

Running from command line "flatpak run com.prusa.PrusaSlicer" yields errors relating to "WebViewDialog error: wxWEBVIEW_NAV_ERR_OTHER" but I don't know if these are truly related to the problem or not.  While I haven't found anything online that seems to be related, one comment suggested overriding WEBKIT_DISABLE_COMPOSTING_MODE=1 in a vaguely related issue from some time ago, but naturally that didn't affect this problem.

### Checklist of files included above

- [x] Project file
- [x] Screenshot

### Version of PrusaSlicer

Version 2.9.1

### Operating system

Ubuntu 24.04.2 LTS

### Printer model

Mk4S, Mk3.9S, Mk2.5S

evancharlton, 2025-05-23:
I started hitting this earlier this week. Sometimes the dialog would flicker something about an unrecognized TLS certificate. Restarting the app made this go away, but I can't use the Send to Connect button.

I'll accept any debug tips anyone might have :)

Jan-Soustruznik, 2025-05-28:
Hello @carltondhouston, @evancharlton, Thank you for your report. Could you please send us the logs when the issue occurs? Try with the latest release (2.9.2), please.

According to the Flatpak installation method, please run the following command in your terminal:
 `flatpak run com.prusa3d.PrusaSlicer//stable --loglevel=4` OR `flatpak  --user run com.prusa3d.PrusaSlicer//stable --loglevel=4` according to method installation of flatpak 

Then, copy the output and send it to us as a .txt file.

Thank you for your help, and have a great day

evancharlton, 2025-05-30:
Just to clarify: this is a regression - this version of PrusaSlicer (2.9.2-rc2 & before) have worked on a long-running basis since .. I don't know, at least January?

Multiple components have changed over time, so I can't say it's **definitely** an issue on the service side of things, but I just thought I'd point it out.

thefnordling, 2026-01-19:
my kid asked me to print something for him today - but i couldn't log into prusa slicer!  i am on ubuntu 24.04, i was trying to use prusa slicer 2.9.4 and was having no luck at all with the flatpak install (was getting the error shown in this issue/report).   It was a fresh install for me and i was hitting the issue during the config wizard step of first launch.  I tried using both google and my prusa account/password to log in and both were failing.

After cloning this repository and building from source i was able to reproduce the issue, so sort of confirmed it's not a flatpak problem (i had originally suspected the issue was a flatpak permission problem).

dug into it a bit with claude - it looks like some non-critical errors in the login page were causing wxWEBVIEW_NAV_ERR_OTHER to be generated, and the error handler was treating all of those errors as critical and was interrupting the login process.   after relaxing the error handling i was able to log in.  

there were some odd interactions with the google login appearing in both external browser window (new popup) as well as within the slicer ui - but with the fix i am now able to log in and can access/manage my printer via prusa connect from prusa slicer (i'm saved) on my linux workstation.

i'm not familiar with the codebase, and this is a change that impacts  login process, so given the potential security impacts i wouldn't expect my PR to be merged but @Jan-Soustruznik i will open a PR with the changes that worked for me (and will tag you as well as this issue issue) in case it saves time for someone who's familiar enough with the codebase to safely make changes to the login page.

closed 2026-09-29 by prusaslicer-bot[bot]