prusa3d/PrusaSlicer #14357: 2nd Layer Issue when painting cut extruded faces or boolean difference objects - Incorrect Color - Verified not just visual issue, it prints this way

nointernetforyou, 2025-03-25:
### Description of the bug

When painting a relief cut, cut extrude or boolean difference the 2nd layer up inside the cut is incorrect. I have adjusted all things in Slicer, I have made .stp and .stl files in Solidworks, Blender, F360 and it is always the same result.

Please help, I really need this to work for a project and the only work around makes the printing 20% longer and x4 more tool changes.

Version 2.9.1

[image]

[image]

[image]

[image]

### Project file & How to reproduce

[Bug_Report_Layer_Issue.zip](https://github.com/user-attachments/files/19440242/Bug_Report_Layer_Issue.zip)

### Checklist of files included above

- [x] Project file
- [x] Screenshot

### Version of PrusaSlicer

2.9.1

### Operating system

Windows 10 Pro

### Printer model

XL5

u89djt, 2025-03-25:
(From a fellow user) What you're seeing is the main object colour visible in the corner in a gap between the horizontal coloured region and the coloured slope. There isn't a main colour layer printed in the area you coloured: the backgound is showing through.
[image]
Certainly, there needs to be a way of closing or colouring that kind of gap appropriately. It might be there but I haven't found it yet.
What's the workaround you mention?
I made explicit colour bodies:

[paintparts.zip](https://github.com/user-attachments/files/19445784/paintparts.zip)

[image]

nointernetforyou, 2025-03-25:
> (From a fellow user) What you're seeing is the main object colour visible in the corner in a gap between the horizontal coloured region and the coloured slope. There isn't a main colour layer [...]

Well I guess I'm not sure how to make explicit color bodies. I primarily use Solidworks, I wonder if this is something in SW.

My work around is to make the parts color blue (in this case) Extruder 3, then paint the outer walls of the part orange, Extruder 1. It works, but it makes a very thin top layer, makes blue layers all the way to Layer 2 off the build plate which causes a lot of tool changes and extra time.

Thank you for showing me it can be done somehow. Prusa should take a look at this though. The gap is might be .00001 somehow, if that and doesn't mean it needs to account for 0.25mm layer height.

u89djt, 2025-03-26:
OK, you can definitely do it in Solidworks. I happened to do it in Fusion, but I think we're in very standard operation territory.
Take a copy of the surfaces of the inside of the arrow and that cylindrical cavity.
Thicken them outwards. You might want to trim the tops down.
Cut those out of the main object bodies.
In the slicer, set the gap closing radius to 0.0
[image]
Separate to parts
Select the right extruders for all those parts.

u89djt, 2025-03-26:
I put a step file in the zip.

closed 2026-09-29 by prusaslicer-bot[bot]