prusa3d/PrusaSlicer #14365: Prusaslicer 2.9.1. closes when I slice.

Peter-MGP, 2025-03-26:
### Description of the bug

Hello,
I'm a new Prusa user (already experienced with banbustudio). I just installed Prusaslicer 2.9.1.

When I start a slice, the software automatically closes when I reach 95% of the slice rendering. I've tried with very small files of different origins, .stl and .3mf, and the result is always the same.

Configuration: HP Z840 Workstation with Windows 10 Pro 64-bit and NVIDIA Quadro M2000

### Project file & How to reproduce

1.open Prusaslicer
2.open an .3mf files
3.push the button to slice (ctrl+6)
4.progretion of calul 90%
5.Prusa slicer shut down

### Checklist of files included above

- [x] Project file
- [ ] Screenshot

### Version of PrusaSlicer

Prusaslicer 2.9.1.

### Operating system

Windows 10 Pro 64-bit

### Printer model

Prusa XL

murk-sy, 2025-03-26:
Can you go to `Help > System info`, click "Copy to Clipboard" and then post that information?

Ideally also provide an example 3mf, even if it's a simple model of a cube, just so it's easier for devs to know exactly what they're dealing with.

Peter-MGP, 2025-03-26:
PrusaSlicer
Version:   2.9.1
Build:     PrusaSlicer-2.9.1

Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 10 (version 19045), édition 64-bit
Total RAM size [MB]: 137,350MB
OpenGL installation
GL version:   4.6.0 (4.6.0 NVIDIA 572.83)
Profile:      Core
Vendor:       NVIDIA Corporation
Renderer:     Quadro M2000/PCIe/SSE2
GLSL version: 4.60.0
Textures compression:       Enabled
Mutisampling: Enabled (4 samples)
<details>
<summary>Installed extensions:</summary>
GL_AMD_multi_draw_indirect
GL_AMD_seamless_cubemap_per_texture
GL_AMD_vertex_shader_layer
GL_AMD_vertex_shader_viewport_index
GL_ARB_ES2_compatibility
GL_ARB_ES3_1_compatibility
GL_ARB_ES3_2_compatibility
GL_ARB_ES3_compatibility
GL_ARB_arrays_of_arrays
GL_ARB_base_instance
GL_ARB_bindless_texture
GL_ARB_blend_func_extended
GL_ARB_buffer_storage
GL_ARB_cl_event
GL_ARB_clear_buffer_object
GL_ARB_clear_texture
GL_ARB_clip_control
GL_ARB_color_buffer_float
GL_ARB_compressed_texture_pixel_storage
GL_ARB_compute_shader
GL_ARB_compute_variable_group_size
GL_ARB_conditional_render_inverted
GL_ARB_conservative_depth
GL_ARB_copy_buffer
GL_ARB_copy_image
GL_ARB_cull_distance
GL_ARB_debug_output
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
GL_ARB_derivative_control
GL_ARB_direct_state_access
GL_ARB_draw_buffers
GL_ARB_draw_buffers_blend
GL_ARB_draw_elements_base_vertex
GL_ARB_draw_indirect
GL_ARB_draw_instanced
GL_ARB_enhanced_layouts
GL_ARB_explicit_attrib_location
GL_ARB_explicit_uniform_location
GL_ARB_fragment_coord_conventions
GL_ARB_fragment_layer_viewport
GL_ARB_fragment_program
GL_ARB_fragment_program_shadow
GL_ARB_fragment_shader
GL_ARB_fragment_shader_interlock
GL_ARB_framebuffer_no_attachments
GL_ARB_framebuffer_object
GL_ARB_framebuffer_sRGB
GL_ARB_geometry_shader4
GL_ARB_get_program_binary
GL_ARB_get_texture_sub_image
[... 314 lines cut ...]
GL_NV_transform_feedback2
GL_NV_uniform_buffer_std430_layout
GL_NV_uniform_buffer_unified_memory
GL_NV_vertex_array_range
GL_NV_vertex_array_range2
GL_NV_vertex_attrib_integer_64bit
GL_NV_vertex_buffer_unified_memory
GL_NV_vertex_program
GL_NV_vertex_program1_1
GL_NV_vertex_program2
GL_NV_vertex_program2_option
GL_NV_vertex_program3
GL_NV_viewport_array2
GL_NV_viewport_swizzle
GL_OVR_multiview
GL_OVR_multiview2
GL_S3_s3tc
GL_SGIS_generate_mipmap
GL_SGIS_texture_lod
GL_SGIX_depth_texture
GL_SGIX_shadow
GL_SUN_slice_accum
GL_WIN_swap_hint
WGL_EXT_swap_control
</details>

Peter-MGP, 2025-03-26:
Unfortunatly this widows refuse to load my .3mf ... (File type not allowed: .3mf)

closed 2026-09-29 by prusaslicer-bot[bot]