prusa3d/PrusaSlicer #14366: mutipul extruder setting for the XL  X5  infill/solid extruder gets over-ridden by coloring x4 the model

raizac, 2025-03-27:
In the slicer tab, under printer settings, in the left pane window, scroll down to multiple extruders. Then, in the main window, set the extruder.

[image]

All the other extruders in the slicer do what they need to do, but the infill extruder and solid infill extruder are overridden when coloring the model with 4 colors.

Non-colored model:

[image]

This is what happens when you color the model with 4 colors, and the 5th is for the support extruder:

[image]

As of now, the perimeter extruder is being used as the infill extruder, even though I have the extruder set to T5. It is being overridden by the colors I have on this model. So, if I have a tri-color filament, I have to use it as both the perimeter and infill extruder.

This is a specific setting for the extruder. It should override anything else in the slicer.

Changing this would allow people to save on special filaments like tri-color or other special colors. This may increase print time, but it would allow for more prints with these special-colored filaments!

raizac, 2025-04-04:
Hi, you should be able to recreate it with any model, just x4 color the model and then try and set the extruder infile to the 5th print head. it will not print it will use the 1st print head as the infill

u89djt, 2025-04-04:
The procedure for bug reporting is to provide a complete example of the problem so that everyone is definitely looking at precisely the same thing. You say it "should" be possible to reproduce the bug. Pick a model that you can post and see if you do get the same problem. If you do, post that.

raizac, 2025-04-04:
sure here is anther model  t5 is infill and YELLOW in the slicer, and as you can see, it is not the infill 

[image]

[image]

Removed all colors 

[image]
it works

hears another one, no color 

[image]

Same setting, now watch this. I have a good one for you  @ 2 different infills 

[image]

[image]

This is the top part 

[image]

seems to be working on the wipe tower as t5 but not infill

u89djt, 2025-04-04:
You didn't post a3mf file.

raizac, 2025-04-04:
u ask me to repoduce it i just did and you still need the 3mf. well here you can have this as it';s my design and you can have the dragon box as well it's on thingavers 

[menu.zip](https://github.com/user-attachments/files/19610903/menu.zip)

[dragon boz for example.zip](https://github.com/user-attachments/files/19610904/dragon.boz.for.example.zip)
not sure if this is the right way of sending 
NO this is not how I paint for the dragon was just a quick paint to show

raizac, 2025-04-04:
Well, I was told by Prusa tech support that this was not a bug, but how the program works, just like my other post in cutting and colors, how the colors get deleted after a cut to the model

 But hey I'm new, not sure how this all works, but I'm willing to learn

u89djt, 2025-04-04:
Good stuff. You can be confident at least one person will take a look - I'm interested, so if it isn't cleared up by anyone else, I'll dig around in case I can see anything. I like concocting workarounds if I can.

u89djt, 2025-04-05:
Right, there's something going on that neither of us can see yet. I cropped and painted the object and got the requested infill. That says to me that either I missed something in your 3mf file, or the slicer tries to obey the infill extruder instruction, but there are circumstances where it can't. 
[dragonpaintok.zip](https://github.com/user-attachments/files/19614912/dragonpaintok.zip)
[image]

u89djt, 2025-04-05:
OK, I see what's happening. You've painted height ranges over the whole object, so none of the volume is the default colour of it's defining mesh, and so it doesn't get the global infill setting. Every volume is special, and is dealt with by the colouring generator.
A colour region created to obey painting is not a standard body: it has no base or top, it just dives into the object horizontally, and it's all extruded with the relevant colour. 
Your workaround would be to cut the object into two parts and set the default colour and infill extruder for each:
[dragonpaintok.zip](https://github.com/user-attachments/files/19615083/dragonpaintok.zip)
[image]
On thing to note is that when I added infill extruder settings to the parts, which were added as 5 without my changing it, the slicer didn't mark the result I tested without setting the infill extruders individually as stale, and I had to change the extruder to something else and back again to make it reslice. Without setting the infill extruders individually, the infill is the same as the perimeters.

raizac, 2025-04-05:
So if I'm getting this right, when I paint the model, don't paint with the default tool, which is tool 1, as if I do, it's overriding the default infill tool that I set 

Not sure if I'm explaining this right 

So I just did a test, as long as I do not paint the entire model with tool 1, 

So these are the steps 
Of course, load the model
Click on paint 
Now, if you left/right click on the model, it will now set that as the main paint for the entire model, overriding the infill setting of t5.
Click Clear all paint
 
Now reset out of paint << just helps to reset the default 
go back into paint, do not left/right click on the model, as that is the default color 
goto brush paint the model  with t2 ,t3,,t4 ,t5  the default infill tool 5 is not over-ridden
You can use Smart Fill and bucket fill as long as you do not have the Smart Fill angle set to X%,* same with bucket fill 

[image] this is set with default t1 and the rest of the is brush on t2 t3 t4 

This is smart infill. The blue is now deep inside the model, and the infill in that area is now painted with that color 

[image]
* When you move the angle of smart fill/bucket fill, the model will highlight with what it will color, the higher the more it will infill and override the infill tool setting 

Thanks for helping I hope I got this right

u89djt, 2025-04-05:
You've found that painting an object whose default tool is t1 with t1 creates a local volume within which the global infill tool setting is not used. That's definitely worth knowing.
"Now, if you left/right click on the model, it will now set that as the main paint for the entire model, overriding the infill setting of t5." doesn't sound right.
I just get that local volume with t1 infill.
[localt1.zip](https://github.com/user-attachments/files/19618413/localt1.zip)
[image]
I just painted the corner - changing the default tool of the object shows the paint:
[image]

raizac, 2025-04-05:
but see you painted with a brush if you try it  with smart or bucket @ fill angel of 90  this will pain the entire model with what ever tool you chose and over ride the printer infill setting tool 5

[image]

u89djt, 2025-04-05:
Ahh, I get you. Yeah, any layer that is painted all the way round  - whether in one override colour or more - will not get a chance to use the global infill setting.
So the question remains: could a global infill tool be applied even in paint fulfilment volumes? Obviously it's possible, but it was chosen against...
The infill colour and structure could show through the perimeters and adversely affect the appearance of painted regions? Safest to fill with the perimeter colour. It might be as simple as that.

raizac, 2025-04-05:
i would think it could as this model im doing right now is 

if the color setting were changed to somthing like this the golbal infill would work

[image]

i think that coloring deep in to the print is uneeded something like this could work. 

this is a menu holder im  making for a mexican restraunt so it's the color of thire flag 
the outer box is just a shield to keep from warping and so far it has work well for me

raizac, 2025-04-05:
heres a pic of it

[image]

and yes i know it's stringy im printing without a print tower as for some reson they are failing mid print

u89djt, 2025-04-06:
The distance between us as human beings being able to see what's possible in an object we believe we understand, and creating algorithms that can find their way to the same conclusions is something that can only be appreciated with some engineering experience. It generally involves assembling a database of scenarios, finding ways of recognizing categories of target that could be approached with a certain processing method, and then fixing what breaks when you introduce that functionality, and learning that you were subtly or dramatically wrong about the category and trying to fix that. And repeat ad infinitum. There's a huge community of magnificently bright and motivated people working on slicing. Just the physics of starting and stopping squirting plastics is still the subject of research, for example.
So for now, you have much of the responsibility for creating a practical scenario for the slicer. In this case, it probably looks like this:
[modmenu.zip](https://github.com/user-attachments/files/19620945/modmenu.zip)
[image]

raizac, 2025-04-06:
WOW wait... umm  Wow—thank you for such a thoughtful and insightful response. That perspective genuinely hit me. I really appreciate the recognition of how nuanced this process is, and the reminder that even defining what we want machines to do is an evolving challenge. I'm learning a lot as I dig deeper into this, and it's encouraging to know that what feels like stumbling around is actually part of the path forward. Thanks again—I’ll keep testing and sharing what I find.

so do you think i/we should put in a Feature Suggestion?? as im all green horn here

u89djt, 2025-04-07:
It's probably worth saying that age isn't the point, it's how many times you've successfully (been) pushed into thinking differently about something you thought you understood, and it's usually several times per phenomenon as your awareness of it and its interactions and dependencies builds.

u89djt, 2025-05-03:
@raizac you all good?

raizac, 2025-05-03:
Yes just been I'm busy with life

raizac, 2025-07-23:
So it's this setting that needs to be changed in order to assign infill to a different extruder and save the special filament. Set it to 0.5 for one wall thickness, with a max of 2 walls. This will create a total of 5 to 6 walls (outer and inner).[image]this is what you will get @ .5
[image] this is with it set @ 2 

[image]

If you set it higher, it creates a gap that gets filled with infill using the outer color, making it look like there are 8 walls. See the photo below for reference. this could all so be good to hide the infill [image]

raizac, 2025-07-23:
@u89djt   look how much filament I’m saving!
The bottom photo uses the normal settings and takes 40g.
With the new settings, the top photo only uses 13g — that’s a 27g savings!
Enough to print another dragon with what’s left over.

closed 2026-09-29 by prusaslicer-bot[bot]