prusa3d/PrusaSlicer #14367: Generated filename to long for Ender printers

baumgartner47, 2025-03-27:
### Description of the bug

When autofilling the filename for a gcode file on the storage card the Slicer uses filenames that are too long, they are not shown in the menu on the printer, when shortened the file appears in the list and can be printed.

### Project file & How to reproduce

- Create file with long filename, for example this: Migthy_Filling_Set_for_9_Dosing_Capsules_Stamp_5h4m_0,20mm_205C_PLA_ENDER3V2.gcode
- Try to print it with Ender V2 from sd-card in printer
- Try it after you shortened it to Migthy.gcode

[Migthy_Filling_Set_for_9_Dosing_Capsules_Top.zip](https://github.com/user-attachments/files/19488459/Migthy_Filling_Set_for_9_Dosing_Capsules_Top.zip)

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

Version 2.9.0

### Operating system

Windows 10, 22h2

### Printer model

Creality Ender 3 V2

neophyl, 2025-03-27:
So what is your suggested fix for this ?  The filename elements can be controlled by the user as they are part of the print profile and can be customised. (Print Settings>Output Options (must be in expert mode to see the correct settings) 
You could remove the usual {input_filename_base} part that is often in the Output Filename Format field.  The actual part filename is also out of the slicer control, its under yours so if the filename is too long for your specific printer then edit it.

You could also use a post processing script to modify your output filename after slicing.  That's already possible to do in the current software.

baumgartner47, 2025-03-28:
Oc I can do that, but it is a hard problem for new people when they export the file to the SD card and see it on the PC in the filestructure of the SD-card but not in the menu on the printer when you use the automatically generated filename. 

I'd put in a check if the name is longer than say 50 characters and at least give the user a prompt with a warnig.

Jan-Soustruznik, 2025-03-28:
Hello, @baumgartner47, thank you for your feedback. However, I believe you should report this issue to your printer's manufacturer, as the problem may be related to their firmware. I don't think adding a warning message to the slicer is the best solution.

closed 2025-03-28 by Jan-Soustruznik

baumgartner47, 2025-03-28:
I still think the slicer should at least warn people that the printer (especially when there is a config file for that modell) can't handle long filenames. It is frustrating because noobs can't figure out what they did wrong. 

Adding multiple stl files to a print is common and the generated filename adds all of the names. Thats unnecessary, even when it does not mean the printer can't handle it.

But in the end I don't give a fuck, I figured it out. You guys seem to know everything, so we will leave it at that.