prusa3d/PrusaSlicer #14374: Improve handling of transition to single wall perimeter with Arachne

Jmlannan, 2025-03-28:
**Is your feature request related to a problem? Please describe.**
Id like to be able to create a feature that requires a wall to go down to a single external perimeter. Arachne currently handles this by splitting the object and bridging between them which leads to poor quality and more seams than are necessary. e.g.

[image]
[image]

This is printed like so with 2 loops and a line between them.
[image]

**Describe the solution you'd like**
Perimeters that can be made continuous should be i.e.
[image]
or
[image]

**Describe how it would work**
I have no idea how Arachne creates perimeters, this may be completely impossible without a ton of work. I suspect it may be a fundamental problem with how Arachne works. I also have not tested that my alternative improves results but in theory 2 seams are better than 4. An option to select between the two modes shown above to hide seams on one side would be nice.

**Describe alternatives you've considered**
Classic perimeter generator + thin walls also handle this poorly. This kind of feature that goes from 2 perimeters to 1 does not currently seem possible.

**Additional context**
Test dimensions for reproducibility:
[image]
Sliced with default prusa mini 0.4 settings on PrusaSlicer 2.9.1 (3/28.2025).

Jmlannan, 2025-03-29:
Also wanted to add that you get a similar issue if you place a hole in a solid object close to an edge that requires it to go to a single perimeter which seems like a more common issue than the scenario I have presented.

Tupson444, 2026-09-19:
still relevant