prusa3d/PrusaSlicer #14380: Fills in holes

Megamannen, 2025-03-30:
### Description of the bug

I have a hollow square, renders correctly, but slicer fills in the hole.
[image]

[image]

### Project file & How to reproduce

[broken.zip](https://github.com/user-attachments/files/19524756/broken.zip)

### Checklist of files included above

- [x] Project file
- [x] Screenshot

### Version of PrusaSlicer

2.9.1

### Operating system

Manjaro

### Printer model

V-Core 3.1

neophyl, 2025-03-30:
You have the slicing mode set to Close Holes.  

[image]

Really common 'issue'.  Do you happen to use a scroll wheel when going down the settings page ?  Its quite easy to change dropdowns while scrolling down a page.

closed 2025-03-30 by rtyr

Megamannen, 2025-03-31:
😢 

[image]