prusa3d/PrusaSlicer #14383: Uable to change brim width

kakemoms, 2025-03-31:
### Description of the bug

I have an issue in Prusaslicer 2.9.1 in that all models (small/large) get a 2mm brim. I can set the brim to 0mm or 10mm, globally or on individual objects, but they still end up with a 2mm brim. I move them around, but to no change. Them brim stays put at 2mm.

When configuring Prusaslicer for the first time, I put in a brim width of 2mm and it seems to have stuck to that value thereafter. There seems to be no way to change it.

[image]

### Project file & How to reproduce

[OverhangTest.zip](https://github.com/user-attachments/files/19530761/OverhangTest.zip)

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.9.1

### Operating system

Windows 11

### Printer model

Creality CR-10s pro v2 with Klipper/Mainsail (which is irrelevant)

neophyl, 2025-03-31:
What you have pictured there is not a brim.  It's a raft.  You have 1 raft layers selected with an expansion of 1.5mm so what you are seeing as a "2mm brim" is the 1.5mm raft expansion.

Your part is being lifted up and printed on a 1 layer high raft.  As such its not touching the bed at all and doesnt get ANY brim.

If I open your project and set raft layers to 0 and slice you get a 10mm brim as expected -

[image]

neophyl, 2025-03-31:
Supports don't get brim, the software is working as intended.    
A brim connects to an object on the bed to provide more surface area, if your object doesn't contact the bed at all then there is zero purpose for a brim.  

You want more surface area on your support raft then change the raft expansion value.

SachCZ, 2025-03-31:
Hello thank you @neophyl, I can confirm it is the intended behavior. Closing.

closed 2025-03-31 by SachCZ