prusa3d/PrusaSlicer #14384: build_win.bat unable to build some dependencies

amatulic, 2025-03-31:
### Description of the bug

Attn: @jschuh - Prusa staff may ignore this, but I don't know where else to report an issue about build_win.bat.

**Problem:** When trying to compile PrusaSlicer 2.9.1 (VS 17 2022, Windows 11) from a clean repo clone using build_win.bat, some dependencies fail to build, even though it worked fine on prior versions of PrusaSlicer (2.8, and I believe also 2.9.0).

After failing to build boost, I ran it again and that worked, but then it aborted with this message:
"OpenVDB could not be found with the bundled find module. You can try to specify the find module location of your OpenVDB installation with the OPENVDB_FIND_MODULE_PATH cache variable."

Going through the manual build process, it appears that OpenVDB got built successfully as a dependency.

### Project file & How to reproduce

**Steps to reproduce**

1. Make sure you have VS 17 2022 installed.
2. Do a fresh clone of the PrusaSlicer 2.9.1 repo.
2. Run build_win.bat
3. Wait until it stops

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.9.1

### Operating system

Windows 11

### Printer model

Prusa MK3S + MMU2S

pillopaolo, 2025-05-12:
@amatulic: did you find a solution?

I have the same issue, both with ver 2.8.0 and the master downloaded 25.04.2025.
Using x64 Native Tools Command Prompt for VS 2022 (2) on Windows 10

-- Could NOT find PkgConfig (missing: PKG_CONFIG_EXECUTABLE)
-- Could NOT find OpenVDB (missing: openvdb) (found suitable version "8.2.0", minimum required is "5.0")
-- OpenVDB ABI Version: 8
-- Performing Test LLVM_HAS_ATOMICS
-- Performing Test LLVM_HAS_ATOMICS - Success

CMake Error at CMakeLists.txt:590 (message):
  OpenVDB could not be found with the bundled find module.  You can try to
  specify the find module location of your OpenVDB installation with the
  OPENVDB_FIND_MODULE_PATH cache variable.

pillopaolo, 2025-05-17:
Problem occurs only when you build dependencies with -DDEP_DEBUG=OFF.
In that case FindOpenVDB.cmake throws an error because detects: "Debug" in CMAKE_CONFIGURATION_TYPES + missing debug libraries

Solved by passing -DCMAKE_CONFIGURATION_TYPES="Release" (default is DebugReleaseMinSizeRelRelWithDebInfo)

amatulic, 2025-05-17:
Thank you @pillopaolo -- I hope the build instruction pages can be updated with this information.

amatulic, 2025-05-18:
Unfortunately, setting -DCMAKE_CONFIGURATION_TYPES="Release" still didn't allow for compilation.

I started over. Deleted my entire local repo. Cloned it again. Started from the beginning, following the instructions for a Windows build. I didn't bother trying to turn off debug, I just followed the instructions.

At the step where I type the command `cmake .. -CMAKE_PREFIX_PATH=<my path>`, it did stuff for a while and then emitted this error message:
````
CMake Error at CMakeLists.txt:376 (find_package):
  By not providing "FindEigen3.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "Eigen3", but
  CMake did not find one.

  Could not find a package configuration file provided by "Eigen3" (requested
  version 3.3.7) with any of the following names:

    Eigen3Config.cmake
    eigen3-config.cmake

  Add the installation prefix of "Eigen3" to CMAKE_PREFIX_PATH or set
  "Eigen3_DIR" to a directory containing one of the above files.  If "Eigen3"
  provides a separate development package or SDK, be sure it has been
  installed.
````

pillopaolo, 2025-05-18:
Did you pay attention at the errors (in red) during the previous step (dependency compile)?
Did you check if Eigen was actually compiled? Results should be in ..\deps\build\builds\Eigen

In my case I had a serious error with Eigen, related to downloaded file hash.
I had to comment 3rd line in ..\deps\+Eigen\Eigen.cmake (#URL_HASH ...)

Above 2 issues costed me about 2 days of troubleshooting ...

amatulic, 2025-05-18:
Hmm. deps\build\builds\Eigen is there but it's empty.

Because the dependency build takes several hours, I started it before bed and it looked like it completed without issues when I got up the next morning. Scrolling back, the only red error message I see has several variants of
`<path>\deps\build\CMakeFiles\<hash>\dep_wxWidgets-mkdir.rule`
... repeated for -udpdate.rule, -configure.rule, -build.rule, -complete.rule

I commented out the line as you suggested (I created a new issue reporting this too) and it looks like Eigen is now being built.

amatulic, 2025-05-18:
Moving on to the step to generate the Visual Studio project file, it no longer gives me an error about missing Eigen. Instead, it now fails with this:
````
CMake Error at CMakeLists.txt:576 (message):
  OpenVDB could not be found with the bundled find module.  You can try to
  specify the find module location of your OpenVDB installation with the
  OPENVDB_FIND_MODULE_PATH cache variable.
````
I don't recall ever having this much trouble building PrusaSlicer from a fresh clone of the repo.

pillopaolo, 2025-05-18:
Dependency compile seems ok (see message "found suitable version 8.2.0").
Same error message as mine, but maybe for a different reason. 
Look at the logic inside ..\\cmake\modules\FindOpenVDB.cmake.
In my case it was the story of -DDEP_DEBUG=OFF, see above. In your case could be something else.

Anyhow, before wasting other time, I suggest:
1) delete the \build directory (NOT \deps\build one) to remove the cache
2) try again.

pillopaolo, 2025-05-18:
Look at the logic inside ..\cmake\modules\FindOpenVDB.cmake.
In particular, it should execute this line "set(OpenVDB_${COMPONENT}_FOUND TRUE)", where COMPONENT=openvdb.
If not, you will get the above error, i.e. "OpenVDB could not be found with the bundled find module".
And the cause might have nothing to do with the "find module location" mentioned by the error message.

amatulic, 2025-05-19:
Unfortunately, deleting \\build and trying again gave the same result.

The line `set(OpenVDB_${COMPONENT}_FOUND TRUE)` occurs in two places, both inside an if() block. It seems like it's failing both checks.

I didn't set -DDEP_DEBUG=OFF because I wanted to make minimal changes, just get it working according to the instructions before I try turning DEP_DEBUG=OFF.

amatulic, 2025-09-02:
@jschuh - trying again with 2.9.3-beta1, script still fails with this message (path names truncated for readability):
> C:\Program Files\Microsoft Visual Studio\2022\Community\MSBuild\Microsoft\VC\v170\Microsoft.CppCommon.targets(254,5):
> error MSB8066: Custom build for 'deps\build\CMakeFiles\e2452a35cb79f7cc982dfeab4c16937f\dep_wxWidgets-mkdir.rule;
>deps\build\CMakeFiles\e2452a35cb79f7cc982dfeab4c16937f\dep_wxWidgets-download.rule;
>deps\build\CMakeFiles\e2452a35cb79f7cc982dfeab4c16937f\dep_wxWidgets-update.rule;
>deps\build\CMakeFiles\e2452a35cb79f7cc982dfeab4c16937f\dep_wxWidgets-patch.rule;
>deps\build\CMakeFiles\e2452a35cb79f7cc982dfeab4c16937f\dep_wxWidgets-configure.rule;
>deps\build\CMakeFiles\e2452a35cb79f7cc982dfeab4c16937f\dep_wxWidgets-build.rule;
>deps\build\CMakeFiles\e2452a35cb79f7cc982dfeab4c16937f\dep_wxWidgets-install.rule;
>deps\build\CMakeFiles\d0a971f5c7096e1d2b3ad76821be3c69\dep_wxWidgets-complete.rule;
>deps\build\CMakeFiles\e1f5415e5744baa96c9ede4ba5849aa0\dep_wxWidgets.rule;
>deps\CMakeLists.txt' exited with code 1. [...\PrusaSlicer\deps\build\dep_wxWidgets.vcxproj]
>
>ERROR: *** Build process failed at deps step. ***

This is after clearing my hard drive of old code, performing a clean `git clone` of the repo, and running the build script.

amatulic, 2025-09-03:
I have also verified that the manual build process for 2.9.3-beta1 fails with the same error as posted in the original description of this ticket:
````
CMake Error at CMakeLists.txt:576 (message):
  OpenVDB could not be found with the bundled find module.  You can try to
  specify the find module location of your OpenVDB installation with the
  OPENVDB_FIND_MODULE_PATH cache variable.
````