prusa3d/PrusaSlicer #14385: Add shortcut F5 for Reload from disk + extra functionality

AaryanL, 2025-03-31:
**Is your feature request related to a problem? Please describe.

When 3d modeling and modifying a model for slicing it requires hundreds of stl refreshing in slicer and using a menu is very inconvenient and slow.

**Describe the solution you'd like

Please add shortcut F5 - Reload All in Prepare window.
Please make it also work in Preview window so that there will be no need to switch to Prepare and back, and make it auto slice upon reload. (Only when in Preview window of course).
2-b. If some particular part is selected make F5 to reload only that one.

CapnBry, 2025-07-16:
Yes please! F5 only works in the "3D Editor View" and not the "Preview" view, which makes it cumbersome to see the preview, edit my model to fix some sort of thickness problem, then have to switch views to reload.

As the person who added the F5 feature to Slic3r, this has bothered me for a long time :-D  I'm not sure how the UI handles the stacking of hotkeys based on view, because other hotkeys that are in the main menu still work so it seems this is possible.

doofmars, 2026-09-26:
Bumping this issue as it is not added yet to Prusa slicer (tested on 3.0.0-alpha12 build from main).
Based on my testing this even degraded as the "F5 only works in the "3D Editor View"" is no longer working.

I would even go one step further:

Prompt to refresh if file changes on disk, in Prepare or Preview view.