prusa3d/PrusaSlicer #14389: PBT Profile for slicer and firmware

katie1348, 2025-04-01:
**Is your feature request related to a problem? Please describe.**
There is no material type in the slicer or the firmware for PBT

**Describe the solution you'd like**
I would like there to be a material preset for PBT, so the material can match the settings.

**Describe how it would work**
It should work like all the other material types.

**Describe alternatives you've considered**
Currently, I have to pretend that it is PETG, which makes setting up more complex. This is because I need to tell the machine and slicer they are using PETG, when they are really using PBT, a different material.

**Additional context**
No other context that seems important.

neophyl, 2025-04-01:
I don't have an XL but I think you are correct that the printer needs to match the type defined in the slicer.  The printer will probably complain if not.  

Creating a profile in the slicer is as easy as just saving the filament profile as a new one with a custom name.  Same as the slicer has been able to do since forever.  Use the disc icon 

[image]

and give it a suitable name when the prompt appears
[image]

You might also want to detach the profile from the system one, depending on what filament profile you use to base it on.

You can save your own profiles for print, filament and printer. 

The XL repository is here https://github.com/prusa3d/Prusa-Firmware-Buddy/issues

although looking at this issue https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/4448  it seems that from firmware version 6.2.0 it is possible to add custom ones to the printer.  So it looks like your feature requests are both already possible to do.

rtyr, 2025-04-02:
Yes, custom filament type can be added on the printer side.

I believe we can close this.

closed 2025-04-02 by rtyr