prusa3d/PrusaSlicer #14392: Ramp after final extrusion

dunkmann00, 2025-04-01:
**Is your feature request related to a problem? Please describe.**
When a print is finished the nozzle just moves up and then parks, this can leave a "zit".

**Describe the solution you'd like**
When ramping lift is enabled, it would be nice if the last move could first ramp away from the part, and then park.

**Describe how it would work**
Rather than just moving the nozzle up, there could be a slight ramp away from the part, enough to avoid any zit, and then the nozzle can move to its parking position.

**Describe alternatives you've considered**
I suppose I could try to manually add a ramping move to one of the `End g-code` custom gcodes. But since I usually always have ramping lift enabled, I would want this final ramp in all my projects, and manually figuring out the position to move to seems like something the slicer could do relatively easily, and something I would probably do slowly and in a highly error prone way.

**Additional context**
I noticed this phenomenon when watching a print finish that was printing 2 of the same part. For the final layer, the first part finished and the nozzle moved over to the next part, the first part's top was smooth. When the second part finished, there was a zit. Just seemed interesting to me that the way the slicer works, both parts didn't have the same top layer finish. When going and checking out the g-code, it was clear the first part ended with a ramp, the second with a vertical travel and then horizontal travel to the park position.

TomaszSzt, 2025-04-18:
(...)I suppose I could try to manually add a ramping move (...)manually figuring out the position to move(...)
Are you aware that you can use "relative move" G-codes? The, if not starting this finishing move at the edge of the build volume, you may be able to get what you need without too much head scratching.

Tupson444, 2025-05-08:
>... "relative move" G-codes

Great idea! I've now added the following at the beginning of end gcode for all of my printers:

```
G91 ; set to relative coordinates
G1 X2 Y2 Z0.15 F5000 ; ramping lift
G90 ; set back to absolute coordinates
```

I've set it to lift at 3° (only 2mm in X and Y direction, so it shouldn't cause collision in any scenario), at ~80mm/s.

The pointy filament ending that I had to cut after every print is now gone.

dunkmann00, 2025-08-06:
The relative moves seem like a reasonable way to achieve this, but I was feeling stubborn and didn't want to use relative moves. I figured out that you can achieve this with absolute positions and also use whatever ramp angle is already set. Replace the first line of the `End G-code` with the following lines:

```
{if layer_z < max_print_height}
G1 X{min(position[0]+10, print_bed_max[0])} Y{min(position[1]+10, print_bed_max[1])} Z{z_offset+min(max_layer_z+(0.25*travel_slope[current_extruder]), max_print_height)} F{travel_speed*60} ; Ramp print head up
{endif}
```

This will move the print head 10mm in the X and Y (towards the back right corner), the Z will be adjusted up at the angle (approximately) that is configured from the printer/filament settings, and the speed is set based on the travel speed from the print settings, which is the usual ramping speed. The `min`s are there just too make sure we don't try to move outside the bounds of the printer.

### No more zits!

---

In case it helps, here is the diff of the entire `End G-code` at the time of this posting, for the MK4S:

```patch
-{if layer_z < max_print_height}G1 Z{z_offset+min(max_layer_z+1, max_print_height)} F720 ; Move print head up{endif}
+{if layer_z < max_print_height}
+G1 X{min(position[0]+10, print_bed_max[0])} Y{min(position[1]+10, print_bed_max[1])} Z{z_offset+min(max_layer_z+(0.25*travel_slope[current_extruder]), max_print_height)} F{travel_speed*60} ; Ramp print head up
+{endif}
M104 S0 ; turn off temperature
M140 S0 ; turn off heatbed
M107 ; turn off fan
G1 X241 Y170 F3600 ; park
{if layer_z < max_print_height}G1 Z{z_offset+min(max_layer_z+23, max_print_height)} F300 ; Move print head up{endif}
G4 ; wait
M572 S0 ; reset PA
M593 X T2 F0 ; disable IS
M593 Y T2 F0 ; disable IS
M84 X Y E ; disable motors
; max_layer_z = [max_layer_z]
```

Tupson444, 2026-09-19:
there is a workaround, but it would still be better if it worked automatically whenever ramping lift is enabled