prusa3d/PrusaSlicer #14394: G-code

funavia, 2025-04-03:
### Description of the bug

I don't know if anyone has written about this before, I don't have time to check, sorry!
But, before, I had to press a SLICE NOW button to create G-code.
Now G-code is created permanently, very often!!!! The Slice Now button has disappeared, and I always have the Export G-code option.
Any modification I make on the Plate, a G-code is created instantly.
But, at the same time, when creating G-code, the type of support I chose is not taken into account.
Is this normal?

### Project file & How to reproduce

It doesn't matter the model, whatever it is, it happens the same way.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

PrusaSIicer 2.9.1

### Operating system

Windows 11

### Printer model

Prusa MK3S

SachCZ, 2025-04-03:
Hello, you probably accidentally turned on `Background processing` in `Preferences` -> `General`. Try to check that and have a nice day.

funavia, 2025-04-03:
Great ! Many Thanks ! Solved !

funavia, 2025-04-03:
[image]

I don't want to waste anyone's time! , but if there is anything else to change here, I would be very grateful!
I am a beginner in this field.

closed 2025-04-03 by SachCZ

TerPuc, 2025-04-04:
Dear @funavia, you can also check [customer support](https://help.prusa3d.com/article/customer-support_2287) and [forum](https://forum.prusa3d.com/) for some tips and tricks from community.