prusa3d/PrusaSlicer #14397: Choose a printer dialog can't be resized

MoritzT, 2025-04-04:
### Description of the bug

On an ultra wide screen, the "choose a printer" dialog when sending to connect is super wide and can't be resized.
There is no need for it to be so wide.

[image]

### Project file & How to reproduce

On a ultra wide screen (5120x1440):
Open any file to print
send to connect
observe "choose a printer" dialog

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.9.1

### Operating system

Windows 11

### Printer model

MK4S

Jan-Soustruznik, 2025-04-04:
Hello @MoritzT, thank you for your report. I've created an internal issue (SPE-2737). We'll aim to address it in one of the upcoming releases.

tkris-sd, 2025-10-06:
It would also be great if it remembered it's size (once you can resize it), or limited itself to the dimensions of the main prusaslicer window when opening. It takes a LOT of mouse movement on ultra-wide screens.

BTW, for me in 2.9.3 it IS resizable, but the minimum size is what appears to be 50% - 50px or something like that. I can make it BIGGER in all directions, but I cannot reduce the width.

closed 2026-09-29 by prusaslicer-bot[bot]