prusa3d/PrusaSlicer #14401: color resets after cut

raizac, 2025-04-04:
Hello, I have Prusa XL 5t 
   I have been working on a 3-color print. It's completely done. I sliced it and noticed that the first layer was off, so I cut it to make the first layer. Once I did this, the color reset to no color at all. I spent a lot of time setting up this color, and I don't understand why it can not just cut the model and leave the color/s!  The cut had nothing to do with the colors 

to be able to copy the multilateral painting 
[image]

have a right-click menu reset/copy

raizac, 2025-04-04:
So I just went back to the other post.  thanks

closed 2025-04-07 by TerPuc

u89djt, 2025-04-07:
In the specific instance of a horizontal cut to remove the first layers, you can move the object in Z to be below the bed instead of cutting, and it retains colour information:
[image]
[image]
If you want to split an object in any single plane with no connectors, and print with the cuts on the bed, you could do it by carefully calculating the rotations and offsets needed:
[image]