prusa3d/PrusaSlicer #14402: Modifier/Part USE SURFACE - gap fill option desired for smooth text/image output

teeny23, 2025-04-04:
I am frustrated by applying Modifiers or Parts as Text or SVG imports on a "rough" or textured surface.

I would like to see an option to add text/svg features to a surface and float the surface gaps with fill if that makes sense.

Please add an option/checkbox to the USE SURFACE area to fill in low spots so that the text or SVG graphic is ultimately smooth but follows the general curvature. 

[image]

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.
I have tried doing it as parts/modifiers, sunken and raised.  Even raised carries the indentions out to the forground

included is a screen shot showing the text following every detail.  It almost needs more of an EMBOSS with the option to set the depth/thickness of the text/svg graphic and how much it should proturde or recess from the general surface ignoring the "cracks" of the pattern.  See the photo of the hockey puck for the desired results (EXCEPT I'd like to do the text in a color and therefore need "anchoring depth")

Jony01, 2025-04-07:
Easist way to do it is add object with surface you want(in your case cylinder) add text on this surface and than delete temporary cylinder surface.

[image]

teeny23, 2025-04-07:
> Easist way to do it is add object with surface you want(in your case cylinder) add text on this surface and than delete temporary cylinder surface.
> 
This is an interesting take... Thank you for taking the time to make a video because I would have never thought you could delete the object you wrapped it around.!

closed 2026-09-29 by prusaslicer-bot[bot]