prusa3d/PrusaSlicer #14404: External perimeter is printed first on thin wall

cobratec, 2025-04-05:
### Description of the bug

External perimeter on inner side is printed before internal perimeters. As there is steep overhang, model fails to print. I have tried differnet combinations of settings wihout any luck. External perimeters first setting is disabled.

[image]

### Project file & How to reproduce

[dispenser.zip](https://github.com/user-attachments/files/19617933/dispenser.zip)

Visible on layers 65 to 241

### Checklist of files included above

- [x] Project file
- [x] Screenshot

### Version of PrusaSlicer

2.9.1

### Operating system

Windows 11

### Printer model

MK4S

lvovalexej, 2025-04-06:
I'm not sure this applies to your case but i've spent a lot of time trying to circumvent wall printing order and found out that it happens when there is only 1 internal perimeter for 2 external walls. as soon as width of the object wall can accommodate 4 line widths or more it prints as intended. This issue persists with Classic wall generator as well as Arachne. 
my workaround is setting internal perimeters to around 90% of the nozzle and tunning Arachne percentages so sloping walls transition from just 2 external walls to 2+2 as soon as possible.
But this is very annoying and i encounter it quite often as i try to print as wide as possible for better strength and speed with wide nozzles and my workaround is adding 1/3 more perimeters slowing printing down.
hope, this helps!

[image]

[image]

[image]

Jan-Soustruznik, 2025-04-07:
Hello, @cobratec, thank you for your report. I created internal issue(SPE-2741). We will try to fix in some of next release. Thank you, have a nice day

cobratec, 2025-04-07:
> I'm not sure this applies to your case but i've spent a lot of time trying to circumvent wall printing order and found out that it happens when there is only 1 internal perimeter for 2 external [...]

Thank you @lvovalexej for your suggestions! I was not able to achieve good results as the wall is too thin in some parts, so I changed the model a little bit to get a thicker wall. With 2 internal perimeters the result is correct.

cobratec, 2025-04-07:
> Hello, [@cobratec](https://github.com/cobratec), thank you for your report. I created internal issue(SPE-2741). We will try to fix in some of next release. Thank you, have a nice day

Thank you!

closed 2026-09-29 by prusaslicer-bot[bot]