prusa3d/PrusaSlicer #14405: "Single Perimeter on Top Surfaces" causes erroneous perimeter intersectingh with top fill lines

ast0815, 2025-04-05:
### Description of the bug

When I enable "Single Perimeter on Top Surfaces" for all surfaces, the slicer creates a weird perimeter going through the centre of a big top surface. It seems completely unnecessary and intersects with the top infill of that layer. I attached a screenshot where this option is enabled for one copy of the model, but not the other:

[image]

It might have something to do with a thin wall that sits on top of that surface? I would expect the slicer to either treat the whole surface as a single top surface, with no internal perimeters, or if it turns it into separate top surfaces, then the infill should not intersect with the resulting perimeters.

System info:

PrusaSlicer
Version:   2.9.1
Build:     PrusaSlicer-2.9.1+flathub.org

Operating System:    Unix
System Architecture: 64 bit
System Version:      Linux 5.15.0-131-generic x86_64
Total RAM size [MB]: 16,406MB
OpenGL installation
GL version:   4.6.0 (4.6 (Core Profile) Mesa 24.3.4 (git-769e51468b))
Profile:      Core
Vendor:       Intel
Renderer:     Mesa Intel(R) Iris(R) Xe Graphics (TGL GT2)
GLSL version: 4.60.0
Textures compression:       Enabled
Mutisampling: Enabled (4 samples)
<details>
<summary>Installed extensions:</summary>
GL_3DFX_texture_compression_FXT1
GL_AMD_conservative_depth
GL_AMD_depth_clamp_separate
GL_AMD_draw_buffers_blend
GL_AMD_gpu_shader_int64
GL_AMD_multi_draw_indirect
GL_AMD_performance_monitor
GL_AMD_pinned_memory
GL_AMD_query_buffer_object
GL_AMD_seamless_cubemap_per_texture
GL_AMD_shader_stencil_export
GL_AMD_shader_trinary_minmax
GL_AMD_texture_texture4
GL_AMD_vertex_shader_layer
GL_AMD_vertex_shader_viewport_index
GL_ANGLE_texture_compression_dxt3
GL_ANGLE_texture_compression_dxt5
GL_ARB_ES2_compatibility
GL_ARB_ES3_1_compatibility
GL_ARB_ES3_2_compatibility
GL_ARB_ES3_compatibility
GL_ARB_arrays_of_arrays
GL_ARB_base_instance
GL_ARB_blend_func_extended
GL_ARB_buffer_storage
GL_ARB_clear_buffer_object
GL_ARB_clear_texture
GL_ARB_clip_control
GL_ARB_compressed_texture_pixel_storage
GL_ARB_compute_shader
GL_ARB_compute_variable_group_size
GL_ARB_conditional_render_inverted
GL_ARB_conservative_depth
GL_ARB_copy_buffer
GL_ARB_copy_image
GL_ARB_cull_distance
GL_ARB_debug_output
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_derivative_control
GL_ARB_direct_state_access
GL_ARB_draw_buffers
GL_ARB_draw_buffers_blend
GL_ARB_draw_elements_base_vertex
GL_ARB_draw_indirect
GL_ARB_draw_instanced
GL_ARB_enhanced_layouts
GL_ARB_explicit_attrib_location
GL_ARB_explicit_uniform_location
GL_ARB_fragment_coord_conventions
GL_ARB_fragment_layer_viewport
GL_ARB_fragment_shader
GL_ARB_fragment_shader_interlock
GL_ARB_framebuffer_no_attachments
GL_ARB_framebuffer_object
GL_ARB_framebuffer_sRGB
GL_ARB_get_program_binary
GL_ARB_get_texture_sub_image
GL_ARB_gl_spirv
GL_ARB_gpu_shader5
GL_ARB_gpu_shader_fp64
GL_ARB_gpu_shader_int64
GL_ARB_half_float_pixel
GL_ARB_half_float_vertex
GL_ARB_indirect_parameters
GL_ARB_instanced_arrays
GL_ARB_internalformat_query
GL_ARB_internalformat_query2
GL_ARB_invalidate_subdata
GL_ARB_map_buffer_alignment
GL_ARB_map_buffer_range
GL_ARB_multi_bind
GL_ARB_multi_draw_indirect
GL_ARB_occlusion_query2
GL_ARB_parallel_shader_compile
GL_ARB_pipeline_statistics_query
GL_ARB_pixel_buffer_object
GL_ARB_point_sprite
GL_ARB_polygon_offset_clamp
GL_ARB_post_depth_coverage
GL_ARB_program_interface_query
GL_ARB_provoking_vertex
GL_ARB_query_buffer_object
GL_ARB_robust_buffer_access_behavior
GL_ARB_robustness
GL_ARB_sample_shading
GL_ARB_sampler_objects
GL_ARB_seamless_cube_map
GL_ARB_seamless_cubemap_per_texture
GL_ARB_separate_shader_objects
GL_ARB_shader_atomic_counter_ops
GL_ARB_shader_atomic_counters
GL_ARB_shader_ballot
GL_ARB_shader_bit_encoding
GL_ARB_shader_clock
GL_ARB_shader_draw_parameters
GL_ARB_shader_group_vote
GL_ARB_shader_image_load_store
GL_ARB_shader_image_size
GL_ARB_shader_objects
GL_ARB_shader_precision
GL_ARB_shader_stencil_export
GL_ARB_shader_storage_buffer_object
GL_ARB_shader_subroutine
GL_ARB_shader_texture_image_samples
GL_ARB_shader_texture_lod
GL_ARB_shader_viewport_layer_array
GL_ARB_shading_language_420pack
GL_ARB_shading_language_include
GL_ARB_shading_language_packing
GL_ARB_spirv_extensions
GL_ARB_stencil_texturing
GL_ARB_sync
GL_ARB_tessellation_shader
GL_ARB_texture_barrier
GL_ARB_texture_border_clamp
GL_ARB_texture_buffer_object
GL_ARB_texture_buffer_object_rgb32
GL_ARB_texture_buffer_range
GL_ARB_texture_compression_bptc
GL_ARB_texture_compression_rgtc
GL_ARB_texture_cube_map_array
GL_ARB_texture_filter_anisotropic
GL_ARB_texture_float
GL_ARB_texture_gather
GL_ARB_texture_mirror_clamp_to_edge
GL_ARB_texture_multisample
GL_ARB_texture_non_power_of_two
GL_ARB_texture_query_levels
GL_ARB_texture_query_lod
GL_ARB_texture_rectangle
GL_ARB_texture_rg
GL_ARB_texture_rgb10_a2ui
GL_ARB_texture_stencil8
GL_ARB_texture_storage
GL_ARB_texture_storage_multisample
GL_ARB_texture_swizzle
GL_ARB_texture_view
GL_ARB_timer_query
GL_ARB_transform_feedback2
GL_ARB_transform_feedback3
GL_ARB_transform_feedback_instanced
GL_ARB_transform_feedback_overflow_query
GL_ARB_uniform_buffer_object
GL_ARB_vertex_array_bgra
GL_ARB_vertex_array_object
GL_ARB_vertex_attrib_64bit
GL_ARB_vertex_attrib_binding
GL_ARB_vertex_buffer_object
GL_ARB_vertex_shader
GL_ARB_vertex_type_10f_11f_11f_rev
GL_ARB_vertex_type_2_10_10_10_rev
GL_ARB_viewport_array
GL_ATI_blend_equation_separate
GL_ATI_texture_float
GL_EXT_EGL_image_storage
GL_EXT_EGL_image_storage_compression
GL_EXT_EGL_sync
GL_EXT_abgr
GL_EXT_blend_equation_separate
GL_EXT_debug_label
GL_EXT_demote_to_helper_invocation
GL_EXT_depth_bounds_test
GL_EXT_draw_buffers2
GL_EXT_draw_instanced
GL_EXT_framebuffer_blit
GL_EXT_framebuffer_multisample
GL_EXT_framebuffer_multisample_blit_scaled
GL_EXT_framebuffer_object
GL_EXT_framebuffer_sRGB
GL_EXT_memory_object
GL_EXT_memory_object_fd
GL_EXT_packed_depth_stencil
GL_EXT_packed_float
GL_EXT_pixel_buffer_object
GL_EXT_polygon_offset_clamp
GL_EXT_provoking_vertex
GL_EXT_semaphore
GL_EXT_semaphore_fd
GL_EXT_shader_framebuffer_fetch
GL_EXT_shader_framebuffer_fetch_non_coherent
GL_EXT_shader_integer_mix
GL_EXT_shader_samples_identical
GL_EXT_texture_array
GL_EXT_texture_compression_dxt1
GL_EXT_texture_compression_rgtc
GL_EXT_texture_compression_s3tc
GL_EXT_texture_filter_anisotropic
GL_EXT_texture_integer
GL_EXT_texture_sRGB
GL_EXT_texture_sRGB_R8
GL_EXT_texture_sRGB_decode
GL_EXT_texture_shadow_lod
GL_EXT_texture_shared_exponent
GL_EXT_texture_snorm
GL_EXT_texture_storage
GL_EXT_texture_swizzle
GL_EXT_timer_query
GL_EXT_transform_feedback
GL_EXT_vertex_array_bgra
GL_EXT_vertex_attrib_64bit
GL_IBM_multimode_draw_arrays
GL_INTEL_blackhole_render
GL_INTEL_conservative_rasterization
GL_INTEL_performance_query
GL_INTEL_shader_atomic_float_minmax
GL_INTEL_shader_integer_functions2
GL_KHR_blend_equation_advanced
GL_KHR_blend_equation_advanced_coherent
GL_KHR_context_flush_control
GL_KHR_debug
GL_KHR_no_error
GL_KHR_parallel_shader_compile
GL_KHR_robust_buffer_access_behavior
GL_KHR_robustness
GL_KHR_texture_compression_astc_ldr
GL_KHR_texture_compression_astc_sliced_3d
GL_MESA_framebuffer_flip_y
GL_MESA_pack_invert
GL_MESA_shader_integer_functions
GL_MESA_texture_const_bandwidth
GL_MESA_texture_signed_rgba
GL_NV_alpha_to_coverage_dither_control
GL_NV_compute_shader_derivatives
GL_NV_conditional_render
GL_NV_copy_image
GL_NV_depth_clamp
GL_NV_fragment_shader_interlock
GL_NV_packed_depth_stencil
GL_NV_texture_barrier
GL_OES_EGL_image
GL_S3_s3tc
</details>

### Project file & How to reproduce

[5inch-blank-Body.zip](https://github.com/user-attachments/files/19618084/5inch-blank-Body.zip)

### Checklist of files included above

- [x] Project file
- [x] Screenshot

### Version of PrusaSlicer

 PrusaSlicer-2.9.1+flathub.org

### Operating system

Ubuntu 20.04

### Printer model

Cetus with TinyFAB V5 CPU

SachCZ, 2025-04-07:
As I see it is is a model issue. Thank you @u89djt. Closing.

ast0815, 2025-04-07:
Thanks for having a look! But I don't quite understand the reasoning of closing this issue. The fact that only 3 out of the 4 cross bars get the perimeters below it, I don't actually care about. The actual bug (to me at least) is that it is doing the top-layer infill pattern right _through_ those perimeters in a single movement. It fills the whole rectangular area completely ignoring the perimeters inside it. That means there would be too much material extruded where it is trying to put both the perimeter and the top layer infill.

u89djt, 2025-04-07:
Oh hey @SachCZ, they're right:
[image]
So there's something bug-like as well as the quirk of fate.
Excellent follow-through, @ast0815

u89djt, 2025-04-09:
OK, emphatically a bug, and less trivially related to object width variation.
This is the original object and settings with the original broken perimeter driven through top fill, then next to it a copy of the same object and settings but with a cylinder part added to it outside it's volume.
[withoutboardcylinder-5inch-blank-Body.zip](https://github.com/user-attachments/files/19666501/withoutboardcylinder-5inch-blank-Body.zip)
[image]
Seems to depend on what order some threshholds are met?

SachCZ, 2025-04-10:
Hello, I see. I previously did not notice the obvious problem. This is a bug.

lukasmatena, 2026-09-01:
Fixed in [3.0.0-alpha11](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_3.0.0-alpha11). Closing.

closed 2026-09-01 by lukasmatena