prusa3d/PrusaSlicer #14407: Failure Prints recovery procedure

Bartnoone, 2025-04-06:
Is your feature request related to a problem? Please describe.
When using Prusa Slicer, the process of recovering a print is extremely cumbersome. Currently, if a problem occurs and part of the print remains on the build plate, the only options are to manually edit the gcode or decompose the 3MF files. Additionally, the introduction of Bcode has further complicated matters, making it nearly impossible to quickly resume printing and leading to significant user frustration.

Describe the solution you'd like.
I would like to see a print recovery feature integrated into Prusa Slicer. The software should allow the user to input the height at which the print remains on the build plate, and then automatically generate a new gcode file that resumes printing from that layer.

Describe how it would work.

The user enters the current print height at which the print has stopped.

Prusa Slicer analyzes the original gcode and generates a new file that starts printing from the specified layer.

The printer then continues the job using the new gcode, eliminating the need for manual code editing and saving valuable time and material.

Describe alternatives you've considered.
The only alternatives available now involve manually editing gcode files or decomposing 3MF files, both of which are time-consuming and error-prone. An automated recovery feature would be a much simpler and more reliable solution.

Additional context.
Implementing this feature would significantly streamline the workflow in Prusa Slicer, reducing user frustration and allowing for more efficient printing. Users want to focus on printing rather than on manual gcode adjustments, making this enhancement essential for improving the overall user experience.

u89djt, 2025-04-06:
(From a fellow user) You can switch bcode off .
[image]
Prusa deal with object cancellation on the printer itself, so I imagine they'll want very specific motivation for going down the route you're suggesting. 
If you can fully describe your printing setup and provide an example where it was difficult for you, someone in a similar situation following the repository might be able to contribute to your argument.
One important barrier to providing the automatic recovery you describe could be in needing a less experienced user to correcly identify where the print failed. If they're out by 0.2mm it's all crashes or doodles in the air. The printer knows where it's up to - mostly.

Bartnoone, 2025-04-06:
Thanks for advise ,Even now, when I look at the gcode, there are so many things in there that you don't know what to cut out so that the printer starts printing from scratch without leveling. The point is that more and more is being put into this gcode and that's why this solution is needed. Yes, but you can't make up for the missing 0.2 mm layer by increasing the flow in the Bamboo lab. You can't do that because there isn't even an option to reduce the flow. I think that a simple step-by-step guide on how to do this is possible. All you need to do is measure the exact height of the model and start printing at too slow a speed when starting the first layer.

u89djt, 2025-04-06:
The developers present are reading "all you need to do" and experiencing quite specific, detailed, expert responses based on knowledge of the myriad little slips a machine and a user can make that lead to spaghetti, crunching noises and support call volume.
The step by step guide for prusa printers is given in the knowledge base.
You haven't said anything about your printing setup except that you've used Bambu lab, which isn't affiliated with Prusa. If you want to influence Bambu lab, that happens somewhere else, not here in the prusaslicer repository.
Since third parties are responsible for third party printing profiles, the development of automated responses to failures on third party machines is going to start with third party effort. I guess - I'm just a user - that collaborative investigation could happen when suggestions from outside start to show the right kind of maturity of functionality.
Your prints are routinely failing, so there may be a problem with your printer or your filament. You haven't shown anything that would enable someone to evaluate your problems.

Bartnoone, 2025-04-06:
I didn't write that I cut into the Bambu lab for Prusa or vice versa. I have a Bambu lab and Prusa and in both cases there is the same problem with this incident. I don't want to find out what happened to the print because at the end of the day you would have to check every centimeter of filament to see if it has the right diameter or measure with a micrometer whether the Prusa printer has chosen the right distance to the table depending on the curvature. I think that this option is necessary for users. I recognize in your text the mind of a software engineer - am I right?

u89djt, 2025-04-06:
What is the incident?

Jan-Soustruznik, 2025-04-07:
Hi @Bartnoone,
Thanks for your feature request. Unfortunately, it's not as simple as telling to slicer "start slicing a model from layer 120." There are many other factors to consider—for example, whether the print failed at the beginning, middle, or end of the layer, and whether the failure occurred on the perimeter, infill, or overhang.

We’ve implemented features that can help to save running prints, such as object cancellation and sequential printing (significantly improved in version 2.9.1). I believe your idea has potential, but it would likely be more feasible in the future, especially once printer hardware is better equipped to support it.

I’ll keep this open as a feature request, but with low priority.
Thank you for your time, and have a great day!

Bartnoone, 2025-04-07:
The way I see it, it's better to have a weaker quality of the lost printout than to have nothing at all, so the solution can be done in this way, that at a given layer height, the printer starts from the beginning of the layer but very slowly, e.g. 5-15 mm/sec, and if necessary, the user increases or decreases the flow rate to compensate for the lack or remove the excess, an experienced user will do well, an inexperienced user should have an over flow of 20-30% as he can determine himself when restoring the printout.  Simply when the tool appears, even if not perfect, the parameters will make it possible to make up for its imperfection. The idea is not to take these files apart, not to invent things like how to bypass the leveling of the table, etc.

closed 2026-09-29 by prusaslicer-bot[bot]