prusa3d/PrusaSlicer #14411: Fails to generate gcode for certain STLs when rotated

degroof, 2025-04-07:
### Description of the bug

The model in question is a lattice grid. When the grid is aligned with the XY axes, slicing completes in a reasonable amount of time. When the grid is rotated 45°, the slicing never completes.

The progress bar gets as far as 90%, CPU usage increases, but slicing does not progress beyond that point.

[image]

### Project file & How to reproduce

Steps to reproduce:

1. Open cs-grid.3mf
2. Hit "Slice now"

[cs-grid.zip](https://github.com/user-attachments/files/19636822/cs-grid.zip)

### Checklist of files included above

- [x] Project file
- [x] Screenshot

### Version of PrusaSlicer

2.9.1

### Operating system

Windows 11 Pro (build 26100.3476)

### Printer model

Prusa MK3S+

u89djt, 2025-04-07:
(Fellow user taking an interest) Confirming that it doesn't seem to terminate at that angle. If I rotate it back to parallel to the axes, I get this error:
[image]
I wonder if it should have thrown that error at the provided angle but didn't and ended up looping round something? Will dig further.

degroof, 2025-04-07:
Odd. Here's a project file with the un-rotated model. Slices just fine.
[cs-grid-OK.zip](https://github.com/user-attachments/files/19639322/cs-grid-OK.zip)

u89djt, 2025-04-07:
I'll have a look that. Just in the meantime: this is almost as neat but easier to manipulate:
[mesh-hack.zip](https://github.com/user-attachments/files/19639342/mesh-hack.zip)
[image]

u89djt, 2025-04-07:
Yes, cs-grid-OK.3mf slices fine for me:
[image]

u89djt, 2025-04-07:
If I deactivate "detect bridging perimeters" in the original cs-grid.3mf, the error after rotating to align and hitting slice goes away. It doesn't allow the slice at 45 degrees to terminate.
[cs-grid-no-bridge-aligned.zip](https://github.com/user-attachments/files/19639495/cs-grid-no-bridge-aligned.zip)
[image]
WIll likely check in tomorrow.

closed 2026-09-29 by prusaslicer-bot[bot]