prusa3d/PrusaSlicer #14413: Brim Color/Brim Extruder

devildoc5, 2025-04-08:
### Description of the bug

Attempting to change color of brim.  Despite what is placed under Print Settings > Multiple Extruders > Extruders > Perimeter Extruder tooltip: "The extruder to use when printing perimeters and brim. First Extruder is 1." Changes to this value do not change the brim color/extruder to anything other than lowest number extruder used on first layer.  Verified with Prusa Support that this does not function as the tooltip states.  Prusa Support asked if I would mind submitting a report on GitHub.

### Project file & How to reproduce

[IndianMotorcyclesLightBox.zip](https://github.com/user-attachments/files/19640683/IndianMotorcyclesLightBox.zip)

Can be achieved by adding any brim to an object.  Brim will automatically come out same color as lowest number extruder used in first layer.  

### Checklist of files included above

- [x] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.9.0

### Operating system

Windows 11

### Printer model

Mk3s+ w/ MMU3, MK4s w/ MMU3

neophyl, 2025-04-08:
Maybe Prusa Support should give better advice and follow their own guidelines for reporting issues, such as checking if an issue already exists before opening new duplicate issues.

Such as for example - 
https://github.com/prusa3d/PrusaSlicer/issues/14179
https://github.com/prusa3d/PrusaSlicer/issues/10491
https://github.com/prusa3d/PrusaSlicer/issues/9067
and possibly the earliest one from September 2021 https://github.com/prusa3d/PrusaSlicer/issues/6979

devildoc5, 2025-04-08:
Wow!  The attitude is strong with this one.  It's not just the inability to change the brim extruder.  It is the fact that the tooltip says this is the option, yet has no effect.  "Bro, do you even read?"

closed 2026-09-29 by prusaslicer-bot[bot]