prusa3d/PrusaSlicer #14414: New printers

Dan263737, 2025-04-08:
Hi, I have been using prusa slicer a long time without missing anything but I just got my new Creality ender 3 V3 SE and I dont see profile for this printer. I know I can make my own preset but official preset will be great. It will be great to see this printers: Ender 3 v3 se, Ender 3 v3 ke, Ender 3 core XZ and Creality Hi. Thank you!

u89djt, 2025-04-08:
Third party printer profiles originate in the user community of that printer. See https://github.com/prusa3d/PrusaSlicer/issues/14272

FidelCapo, 2025-04-09:
Thanks to @u89djt for pointing it out — I can confirm that this is a duplicate of [#14414](https://github.com/prusa3d/PrusaSlicer/issues/14414).
Closing