prusa3d/PrusaSlicer #14416: Fan Speed control when using Ironing

DSpecter92, 2025-04-09:
**Is your feature request related to a problem? Please describe.**
I print fridge magnets and like to use Ironing to make them look better. Since the upgrade from MK4 to MK4S, my magnets keep slightly lifting off the bed due to the part cooling fan being so strong and blowing air to the bottom of the print during Ironing (Ironing is happening at only 4mm height), causing artifacts and in worse case scenarios, layer shifts.

**Describe the solution you'd like**
Adding a print fan speed option in the ironing section should be simple enough (in theory) and would allow situations like mine to be a problem of the past. The whole point of ironing is having the tip hot enough the smooth out the print while extruding tiny bits of plastic to ensure everything is even, so fan speed does not need to be full blown out anyway.

**Describe how it would work**
As per above.

**Describe alternatives you've considered**
I would consider using brim but that would then affect the finish around my prints. I am religiously using IPA to clean my smooth PEI sheet and also washing the plate with soap and warm water. The fan is just too strong!

**Additional context**
Add any other context or paint drawing about the feature request here.

DSpecter92, 2025-04-09:
[image]

[image]

Adding these for context, sent just now to Prusa support due to the exact mentioned issue. You can see it is detaching after the first layer is printed.

u89djt, 2025-04-11:
(From a fellow user) If you can post the 3mf file for that print, someone is likely to think of a workaround, or to get a better feel for the problem. You need to zip the 3mf file.

DSpecter92, 2025-04-12:
> (From a fellow user) If you can post the 3mf file for that print, someone is likely to think of a workaround, or to get a better feel for the problem. You need to zip the 3mf file.

[Project.zip](https://github.com/user-attachments/files/19718144/Project.zip)

The file has ironing on all top surfaces (4mm, 6mm and 8mm).

To avoid this and keep printing magnets, I have now used layer height settings to get rid of the ironing in the 4mm height section. The reported issue disappears at the other two sections (6mm and 8mm) as long as the first one prints correctly (Using single perimeter and an archimedean chord finishing, still looks pretty good to me).

But all this just making me believe even more the lifting is due to the print fan being too strong during ironing.

u89djt, 2025-04-12:
Hardening a hypothesis without attempting to specifically knock it out is a barrier to progress in bug hunting.
The intent of ironing is to smooth the plastic out with heat by moving slowly and depositing a small amount of plastic. The consequence of that is that everything gets hotter. Lingering with a hot tool transmits enough heat to melt a tiny puddle. With insufficient cooling, that creates a big puddle of plastic because it stays hot.
I guess the magnet you're referring to fits in the cavity you've left.
The bridging over the cavity will pull as it cools.
We haven't tested anything yet, so we need to start designing experiments.
A simple one: what happens if you print without the cavity? You need the cavity in the final object, but this might teach us something.

u89djt, 2025-04-14:
You have a perfect storm of contiguous cooling plastic pulling across a gap to relatively isolated towers of plastic with small footprints. Top solid infill is coinciding with the bridging layer. I've included the infill on the bed in bright blue just for emphasis.
You've got an extrusion multiplier of 0.96 set. That might be why the first layer looks gappy in your photo - I assumed the offset hadn't been probed effectively, but that seems like a better explanation, I think. Obviously less squish means less grip.
Have you got a 3mf file that worked on the printer before you upgraded? We can look for differences and see if there's any information about what's happening there.
I haven't found an explanation that would motivate slowing the fan. Printing more slowly overall would likely help if the problem is distortion due to warping and pulling from the bed.
Re cleaning the bed: it took me a while to notice that you can get it to a point where a kitchen towel with isopropyl will grip the bed as you push across it and judder and/or squeak. I'd just been wiping up to that point.
[image]

DSpecter92, 2025-04-14:
The extrusion multiplier was set on that print to 0.96 because I was advised to do so by Prusa support. I forgot to put it back to 1 after. All previous attempts that came up with the issue were done with 1 multiplier, but I do understand that may have caused the gaps on the bottom layer on this specific instance, which obviously may not help with grip.

Attached is a 3mf file of a similar design printed last year, before I upgraded to MK4S. I printed a total of 235 units, without a single issue.

[Early 3mf.zip](https://github.com/user-attachments/files/19740292/Early.3mf.zip)

Can you explain again what you mean regarding cleaning the bed? I didn't quite get it. My routine for keeping the bed clean is using 75% IPA wipes that I buy from Amazon. I use these once per day of printing and that has never failed me in the past. Also maybe every 6 months or so I wash it with dish soap and warm water, dry it immediately with kitchen towel and then wipe it with the same IPA wipes. I am also generally careful to not put my greasy paws all over it.

I understand that cooling is needed for ironing, but the ironing settings doesn't seem to have changed between MK4 and MK4S profiles (that I've noticed, I might be wrong about this). If this fan is significantly stronger than the MK4 iteration, do we really need it to be going at the same percentage of power to achieve the same results?

u89djt, 2025-04-14:
Re cleaning, the PEI bed seems smooth, but it has very fine surface structure which is what the extruded plastic squishes into and holds. Grease builds up within that texture so you have to squish it out, either with alcohol or detergent. When the grease has been removed, the empty texture will grab at the cleaning cloth: when you press while wiping, it'll resist motion and judder. If you haven't got to that point, there's still grease in the texture that will limit incursion of the plastic.

The "Early" project file has bigger contiguous regions of plate contact around the cavity, so those will grip better. Try printing exactly the same object? So far we're looking to test "Project" without a cavity, and "Early".

Stronger fans that get more air to pass over the plastic that is being laid enables faster print speeds. It's directed just at that top region that needs to be cooled as soon as possible. When you hear it whistling as the tool passes a cavity, that's like blowing across the top of a bottle. There's no substantial lateral force being applied to the object overall.

[image]

DSpecter92, 2025-04-16:
Apologies I imported the wrong "early" file. I did print this file with no issue back in November 2024 (before I upgraded to MK4S).

[Early Sto Aleixo.zip](https://github.com/user-attachments/files/19781869/Early.Sto.Aleixo.zip)

Interesting notes on the bed, wasn't aware of those details but by that account the bed is gripping fine then because I feel the resistance when cleaning it. I should be able to run these tests next week, currently got an order to fulfil.

u89djt, 2025-04-16:
Have a look at how much slower the successful print was. That will have cooled the plastic more effectively as the layers progressed. Try printing that slowly. Obviously faster printing is desirable, but when you give the printer a big solid surface over an opening between small islands, it's going to pull away.
[image]

DSpecter92, 2025-04-17:
OK that's fair and I get your points. But counteracting that was a much less powerful fan too, without 360 degree cooling. Wouldn't that have caused an issue before then? Or are print speeds much more important in this aspect?

u89djt, 2025-04-17:
I'm sure it's a factor, but proportion of effects can be totally mesed up by structure. Remember the huge solid raft. The interacting phenomena are simultaneous and fiddly. We both need to be aiming to discover what we're wrong about. What's left at the end is correct. There are infinitely more wrong ideas than right ideas, so we need to get on with evoking and testing. So, print it slowly with your impressive new fan and see what happens. If I'm wrong, I'll benefit from knowing and we can search more. If I happen to be right already,  you've got your answer. Can't know till we try. The point is to find things to try that might tell us something.

closed 2026-09-29 by prusaslicer-bot[bot]