prusa3d/PrusaSlicer #14417: Vase Mode + Sequential Printing = 😀

mkoistinen, 2025-04-09:
**Is your feature request related to a problem? Please describe.**
I would like to do `vase mode` AND `sequential printing` at the same time.

**Describe the solution you'd like**
We should be able to enable vase mode and sequential printing together.

**Describe how it would work**
Sequential printing would work precisely the same as non-vase mode objects; one at a time.

u89djt, 2025-04-09:
(Fellow user with good news for you)
You can already do this - you've missed a setting or something. Here's an example project with sequential vase mode:
[Shape-Box.zip](https://github.com/user-attachments/files/19674743/Shape-Box.zip)
[image]
You can close the issue yourself.
Best wishes.

mkoistinen, 2025-04-09:
Yes. Thank you. I don't know why it didn't want to work before.

closed 2025-04-09 by mkoistinen