prusa3d/PrusaSlicer #14418: Ability to remove incorrect usernames in Ogin Autofill

Shootingblanks00, 2025-04-09:
**Is your feature request related to a problem? Please describe.**
Cannot remove incorrect usernames in the Login Auto-fill drop down box

**Describe the solution you'd like**
To be able to remove incorrect usernames by holding down "Shift" and pressing "Delete" key.

**Describe how it would work**
Click in the username box then use the arrow Down Arrow Key to highlight the incorrect name, hold down "Shift" the press "Delete".

**Describe alternatives you've considered**
Tried holding down "Ctrl" and "Delete", tried holding down "Shift" and pressing "Delete" key, tried holding down "Alt" and pressing "Delete" key.  Tried these on both sides of keyboard,

**Additional context**
When I highlight one of the incorrect names, a message pops up that *Includes data you filled online that was autosaved in Microsoft Edge". I have never used Edge for logging into Printables or any other Prusa website. As a matter of fact, I refuse to use Edge for anything.

Shootingblanks00, 2025-04-10:
w0w

TerPuc, 2025-04-14:
Hi @Shootingblanks00, I assume you are using Windows. In that case the way how to delete such informations is to delete webview folder where these informations are stored. The path to the folder is AppData >Local > PrusaSlicer. The name of the folder should be "EBWebView". I hope this helps!

Shootingblanks00, 2025-04-15:
Thank you! That worked all of the names are gone!

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virus-free.www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

TerPuc, 2025-04-15:
Great to hear! I will close this issue as completed.

closed 2025-04-15 by TerPuc