prusa3d/PrusaSlicer #14420: Проблемы с нарезкой модели.

Fizorg91, 2025-04-10:
### Description of the bug

[image]
[image]
[image]

### Project file & How to reproduce

[1.zip](https://github.com/user-attachments/files/19679976/1.zip)

### Checklist of files included above

- [x] Project file
- [x] Screenshot

### Version of PrusaSlicer

2.9.1

### Operating system

Win10

### Printer model

Ender 3

FidelCapo, 2025-04-10:
Thank you for the report. However, the ZIP file you provided contains only the G-code.

To help us identify the cause of the issue, we need the original project file (.3mf) saved from PrusaSlicer, which includes the model, print settings, and other relevant data.

Could you please zip and upload the .3mf project file where the issue can be reproduced? That would allow us to properly investigate.

Thank you.

hejllukas, 2025-04-10:
Hi, we were able to reproduce your issue and it is a bug.
I created an internal issue, no. SPE-2744.

closed 2026-09-29 by prusaslicer-bot[bot]