prusa3d/PrusaSlicer #14421: PrusaSlicer Crashes (Quits) when slicing this model with 0.20mm but works when slicing 0.15mm

AxionCDM, 2025-04-10:
### Description of the bug

Using the default settings I try to slice this model. When using any 0.20mm slicing method (STRUCTURAL or SPEED) PrusaSlicer closes, however when using 0.15mm SPEED it behaves normally.

### Project file & How to reproduce

[Pampered Chef Measuring Spoon Tray.3mf.zip](https://github.com/user-attachments/files/19683843/Pampered.Chef.Measuring.Spoon.Tray.3mf.zip)

### Checklist of files included above

- [x] Project file
- [ ] Screenshot

### Version of PrusaSlicer

Linux Version 2.9.1 (Flatpak)

### Operating system

Pop!_OS 22.04 LTS

### Printer model

Prusa MK4

FidelCapo, 2025-04-10:
Thank you for the report — we can confirm that the crash occurs as described.

We’ve logged the issue internally under SPE-2747. The problem originates in the Arachne infill engine.

As a temporary workaround, you can switch Arachne to Classic mode in your print settings:
Print Settings → Layers and perimeters → Advanced → Perimeter generator → Classic

We appreciate your help in identifying this issue. We’ll continue investigating.

henrikkoch, 2025-06-24:
Was not helphing on my isseu. I have issue with the last 3 NVIDIA drivers for my M4000 graphic card. They all crashes when slicing. Using the driver before does not?

2tefan, 2026-09-06:
Hi :) I (fellow PrusaSlicer user) had the same issue and did a fix in #15774. With it, I was able to slice your model. Hopefully we'll get it merged, then this issue should be gone

closed 2026-11-02 by lukasmatena