prusa3d/PrusaSlicer #14424: Multi-Material Dual Wipe Towers

lochowa, 2025-04-11:
**Is your feature request related to a problem? Please describe.**
This is related to quality issues regarding Multi-material Printing (Think PETG Supports for PLA)

**Describe the solution you'd like**
Add an option to generate two separate wipe towers, one for each material.  On the XL, maybe make it possible for one tower placed on different heating section so that it's bed adhesion is correct.

**Describe how it would work**
The wipe tower on multi-material prints like PLA/PETG causes the nozzle head to foul quickly, and ruins print quality. If you were able to print separate towers for different material types, you wouldn't have this issue.

**Describe alternatives you've considered**
Even when setting the "Wipe Tower Extruder" option so that the outside of the tower is one material, there are significant enough adhesion issues with the wipe tower that it is difficult to get clean prints because one filament or another fails to stick, the nozzle gets dirty quickly and prints are of poor quality because of this.

Gik-0101, 2025-04-11:
+1

Absolute agree!
Would help immensely to print cleanly with materials that do not stick together.

u89djt, 2025-04-11:
Previous reports on the subject: https://github.com/prusa3d/PrusaSlicer/issues/13161 https://github.com/prusa3d/PrusaSlicer/issues/13566
Bed temperature variation is bad news: https://github.com/prusa3d/PrusaSlicer/issues/11682

Jan-Soustruznik, 2025-04-14:
Issue closed as duplicate of #13161