prusa3d/PrusaSlicer #14425: PrusaSlicer 2.9.2 - Multiple Beds, Send All to Connect - Print time for bed1 set to all beds

brianirwin, 2025-04-11:
### Description of the bug

I create a simple design that has two build plate specified, The first reports 6 hrs 3 minutes and the second is 3 hrs 40. 

The filenames that PrusaSlicer generates often includes the same print-time for both build plates. 

[image]

### Project file & How to reproduce

Create a design with multiple beds
Slice All
Send All to Connect
Check filenames

### Checklist of files included above

- [ ] Project file
- [x] Screenshot

### Version of PrusaSlicer

2.9.

### Operating system

Windows 11

### Printer model

Prusa XL 5T

brianirwin, 2025-05-01:
I have been trying to reproduce without success. Apologies that this does not seem to be a broad issue.

FidelCapo, 2025-05-05:
I’m closing this issue for now as we cannot reproduce it.

However, if you manage to reproduce the problem in the future, please feel free to reopen this issue or add a comment — we’ll take another look.

Thank you!

closed 2025-05-05 by FidelCapo