prusa3d/PrusaSlicer #14427: camera should orbit around selected plate

gdelpu, 2025-04-12:
**Is your feature request related to a problem? Please describe.**
When my project has multiple beds, the view orbits around the first plate created, and not around the selected one

**Describe the solution you'd like**
orbit center for the camera should be refined when selecting a plate

**Describe how it would work**
When I orbit the camera around the objects, I prefer to orbit around the selected plate

**Additional context**
I'm using the Prusa slicer release 2.9.0

neophyl, 2025-04-12:
Essentially a duplicate of the much older report https://github.com/prusa3d/PrusaSlicer/issues/9083
or this one that was closed for example https://github.com/prusa3d/PrusaSlicer/issues/14220
or https://github.com/prusa3d/PrusaSlicer/issues/12323

Basically this is a long running issue as you can see.  You are supposed to select an object and use Z, that then becomes the point of rotation.  However even with that it still has a few niggles, which you will see mentioned if you read the already open (and some closed) existing issues.

TerPuc, 2025-04-14:
Closing as duplicate.

closed 2025-04-14 by TerPuc