prusa3d/PrusaSlicer #14428: Vase mode creates perimeter glitches on fine details

imitation, 2025-04-12:
### Description of the bug

[image]
Vase mode creates what looks like skipped g-code segments when trying to reproduce fine details. The briefly shown (yellow) path preview shows the perimeter generating correctly, but once the preview switches to the detailed g-code preview it includes visible errors.
Slicing the same file with vase mode not enabled produces a clean output.

Solutions tried:
Varying slicing and g-code resolution (no change)
Changing text emboss size (in CAD) from 0.15mm to 0.1mm (no change)
Enabling fuzzy skin (fixes issue at excessive settings)
Disabling arc fitting (no change)

### Project file & How to reproduce

[rakuraku0.15.zip](https://github.com/user-attachments/files/19718963/rakuraku0.15.zip)
3mf project already set up to reproduce issue. Only changes from stock profiles are changing to vase mode and accepting the suggested changes.

### Checklist of files included above

- [x] Project file
- [x] Screenshot

### Version of PrusaSlicer

2.9.2

### Operating system

Win11Pro 24H2

### Printer model

CORE One, XL, ....

u89djt, 2025-04-12:
(From a fellow user) I had a go at working out what was happening and came up with a workaround when this was raised previously: https://github.com/prusa3d/PrusaSlicer/issues/12530

TerPuc, 2025-04-14:
Closing as duplicate.

closed 2025-04-14 by TerPuc