prusa3d/PrusaSlicer #14434: Combine infill feature only works if both infill extruder parameters are equal

BaconFase, 2025-04-12:
### Description of the bug

### Usecase:
Any print where the top and bottom of a model is expected to use the same extruder(material/color) as the side perimeters (So bascially most prints) that would also like to leverage the infill combination feature.

### Problem:
The combine infill feature does not work as expected if **Infill extruder** and **Solid infill extruder** are not set to the same extruder.

### Fix:
Decouple the Solid infill extruder from the infill combination feature.

Solid infill is [usually] the top and bottom of a print which is expected to be the same as perimeters anyway. Infill combination feature is expected to only affect the inner-supporting structure, not anything outwardly visible.

### Project file & How to reproduce

[EXAMPLE Multi Diameter Height Material Print.zip](https://github.com/user-attachments/files/19722191/EXAMPLE.Multi.Diameter.Height.Material.Print.zip)

Set project to combine infill every 3 layers (.6mm layer height infill):
[image]

When the two infill parameters **do not** match each other, the infill stays printed with .2mm layer height:
[image]

When the two infill parameters **do** match each other,  infill is printed with expected .6mm layer height:
[image]

### Checklist of files included above

- [x] Project file
- [x] Screenshot

### Version of PrusaSlicer

2.9.2

### Operating system

Windows 10 Pro

### Printer model

Prusa XL-5T

BaconFase, 2025-04-12:
Forgot to add that my current workaround is to paint both the tops and bottoms of models to the extruder of the perimeters.

For the example with just a couple boxes, this is easy and simple. 
Not so simple and easy the more complex a model is.

jimmysun2000, 2025-05-17:
It looks like the maximum height infill combination can do is limited by the smaller of the two extruders (infill & solid infill). Which means even if you have a 0.6mm nozzle for infill, it will only combine up to 0.2mm, which is your solid infill nozzle. 

I have created a pull request #14566 that decouples solid infill from the infill combination feature, because solid infills are usually on the outside of the print as you described. 

[image]

closed 2026-09-29 by prusaslicer-bot[bot]