prusa3d/PrusaSlicer #14435: Add "Do not slow down for outer walls" for consistent outer perimeter finishes

AK-Chaos, 2025-04-12:
Inconsistent outer wall speeds causing some finish issues, typically on silks, due to slowing outer walls for whatever reason. Orcaslicer has the feature under the filament profile to "do not slow for outer walls" and this makes/keeps outer perimeter finishes incredibly consistent. 

[image]

Just setting maximum outer wall speed, such as PS does, doesn't lead to as consistent of a finish (even with everything else the same) as Orcaslicer. 

I do/have used OS on other printers (Bambu) and my XL, but the XL works MUCH better with PS - but this feature would be a godsend.

rainer1146, 2025-04-18:
I was just about to suggest this great feature of the Orca Slicer myself.
Even better that someone else has already done it.
I definitely support this request.
It's a very useful tool, especially in the fight against VFA.

dwbmb, 2025-04-18:
Count me in for this feature request! (Getting mad printing silk right now)

Tupson444, 2026-09-19:
still relevant