prusa3d/PrusaSlicer #14441: Paint color by layers option

Dmalin3D, 2025-04-14:
**Is your feature request related to a problem? Please describe.**
I have found a few situations where, with the MMU3, in prusaslicer, it would be much more efficient for me to change colors of a selection of layers instead of painting a whole bunch of little bits on a complicated object. Right now, prusaslicer doesn't have a natural option to do with the MMU3. I find the current best option I have is to go back to old school and just at a color change post initial slice and print without the MMU altogether. 

**Describe the solution you'd like**
An option to "paint" layers.

**Describe how it would work**
It probably would be best to have the functionality built into the "Multi-material painting" option but even if it just got built in somehow to the layer height modifier, that would work. 

**Describe alternatives you've considered**
Right now, my options are to go out of the MMU or to do something like a custom gcode fix. 

**Additional context**
So, this would be an example of a file where this would be hugely convenient. I would like to have the hex sections one color and the chain another. It's just one color change but as far as I can tell, there is no natural/efficient solution to this currently in Prusaslicer with the MMU enabled. 

[image]

Dmalin3D, 2025-04-14:
P.S. I did just find a layer height change on 2.9.1. 

I am not yet sure if it's exactly what I want but it may do the trick.

neophyl, 2025-04-14:
Just add a height range modifier to your part (or multiple ones) and then just change the extruder.

[image]

SachCZ, 2025-04-14:
Thanks @neophyl, closing.

closed 2025-04-14 by SachCZ