prusa3d/PrusaSlicer #14446: Can't identify parameter signaled as having an illegal value - localization issue but not just that

nesdnuma, 2025-04-16:
### Description of the bug

Hello,

Some parameter became illegal in PrusaSlicer after setting a nozzle to 0.1mm through Connect on one tool of the XL (this reflects reality as I have the V6 adapter and a .1 nozzle).

[image]

However as I am using the French language in PrusaSlicer, I don't know which parameter the issue is linked to.

Besides the language issue, may I suggest to allow the user to click on the error message and then set the focus of the GUI on the parameter having the wrong value? And maybe set the background color around this parameter in a special color as long as the value is illegal?

### Project file & How to reproduce

[PrusaSlicer config buse 0_1.zip](https://github.com/user-attachments/files/19774511/PrusaSlicer.config.buse.0_1.zip)

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.9.2

### Operating system

Windows 10

### Printer model

Prusa XL 2T

SachCZ, 2025-04-16:
Hello, thank you. I see this both as a localization bug and a feature request. It makes sense, we will try to address it in the future. Thank you.

nesdnuma, 2026-09-18:
This is even worse in 3.0.0. alpha because now one cannot even go below a .25mm nozzle!

[image]