prusa3d/PrusaSlicer #14449: Download path resetting after each restart

Maltesaa, 2025-04-16:
### Description of the bug

The download path set in Preferences > Other gets reset after restarting PrusaSlicer.

### Project file & How to reproduce

1. Go to Configuration > Preferences > Other
4. Set a custom Download path
5. Restart PrusaSlicer
6. See that the download path gets reset

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

Version 2.9.2

### Operating system

Arch Linux

### Printer model

Sovol SV06 Plus

SachCZ, 2025-04-16:
Hi, it does not seem to even be set after closing the preferences window. This is a bug.

SachCZ, 2025-04-16:
SPE-2752

Maltesaa, 2026-09-19:
Issue seems to be resolved as of version 3.0.0-alpha11

closed 2026-09-19 by Maltesaa