prusa3d/PrusaSlicer #14450: Instances of a part are difficult to cancel using the printer UI

ftsabrov, 2025-04-16:
**Is your feature request related to a problem? Please describe.**
When a large amount of instances are added to a bed, it becomes difficult to identify which instance is which, as the arrange algorithm does not consider instance number. This is especially frustrating when one instance must be cancelled via the printer UI, which only lists instance numbers and no visual representation. Instances currently cannot be manually renamed.

**Describe the solution you'd like**
Add an option to the "arrange options" menu that will renumber instances based on the arrange result.

**Describe how it would work**
The arrangement may need to be restricted to lines or a grid. Then, the instances can be numbered first left to right, then back to front.

**Describe alternatives you've considered**
Allowing the user to re-number instances with a text field or a drag-and-drop motion.

**Additional context**
N/A

Tupson444, 2026-09-19:
still relevant

Jan-Soustruznik, 2026-10-21:
A similar feature request already exists. This request has been merge to keep topics organized.

closed 2026-10-21 by Jan-Soustruznik