prusa3d/PrusaSlicer #14451: Abnormal color line in multimaterial slicing

Thonel827, 2025-04-16:
### Description of the bug

Abnormal line are appearing in slicing since couple weeks ago.

[image]

[image]

Tested in 2.9.2 and 2.9.1

Same STL and parameters (0.2mm SPEED / 10% infill) used 3 weeks ago without issues. (Think it was 2.9.1)

[image]

### Project file & How to reproduce

Go to : Slice now

[a3clrc_straight_long_road_curb_line_racing_circuit_complete.zip](https://github.com/user-attachments/files/19784909/a3clrc_straight_long_road_curb_line_racing_circuit_complete.zip)

### Checklist of files included above

- [x] Project file
- [x] Screenshot

### Version of PrusaSlicer

2.9.2

### Operating system

Windows 11

### Printer model

M4KS - MMU3

u89djt, 2025-04-17:
(Fellow user taking an interest)
Top left is your original project, top right is me repainting it to see if I do anything different (I don't seem to), bottom left is with gap closing set to 0.0 to look for glitches in the mesh showing a couple of unintended channels, and bottom right is what the author of the object probably intended: split to parts and assign tools.
I don't know what's causing the odd lines of colour, but it's failing to maintain a coherent picture through the slope on the other side (second image). Maybe there are other voids of overlaps in the mesh that we haven't found yet.
[image]
[image]

Thonel827, 2025-04-17:
Hi ! Thanks for your feedback and time !

Regarding the painting method, I am using straight "bucket fill" on what I want to paint. Quick and effective for what I need (at least, it used to be).

I do not think it is an issue with the model : it is quite a simple model and I used same STL couple weeks ago, it used to work just fine. 
BGCODE recovered from Connect to show it and print correctly - I usualy do not save the .3MF so I miss it. Point of learning to save it for further experiences. 
(Model from Michal Fanta - Printables : https://www.printables.com/model/789942-racing-circuits-lets-build-all-formula-1-grand-pri )

File/Parameter used to work and today is showing systematically the anomaly. Only difference I recall between both is:
 - update from 2.9.1 to 2.9.2 (Tried again with 2.9.1 and same issue appearing).
 - update of presets/configuration (Tried to reactivate to my first configuration at MMU assembly and I still have this anomaly).

Method split part and affect an extruder per parts seems to be working fine though ! Still it feels overkill for the model.

_-- I just noticed I shared 3MF with 0.4HF Nozzle - 0.28mm DRAFT (I tried different configuration/parameters to see if I can find one influencing it), please note I usually print in 0.4 Nozzle - 0.20mm SPEED --_

u89djt, 2025-04-17:
This is a great place to learn by doing free of the stress of your own deadlines :)
It may be a simple model in concept, but the mesh is wobbly on the smallest scale. You've seen that it has gaps. When there are gaps, there are overlaps elsewhere, and they will carry attributes where they shouldn't go, I imagine.
Splitting to parts isn't overkill when it was the intent of the author :)
If you're getting a different result using the same version as before, you're doing something differently. Are you sure it's the same stl?
Here's one hole in the mesh at the front revealed by removing the paint and settiing the join distance to 0.0.
[image]

u89djt, 2025-04-17:
The text is revealing. You'll see that it's a meaningless jumble due to misaligned vertices:
[nopaintzerojoin.zip](https://github.com/user-attachments/files/19798746/nopaintzerojoin.zip)
[image]
With properly aligned vertices, it's a different story:
[test.zip](https://github.com/user-attachments/files/19798755/test.zip)
No gaps or overlaps.
[image]
[image]

u89djt, 2025-04-17:
Right, if rotating it a bit fixes the glitch, there's an implementation detail that can't cope with the mesh the way you'd hope. But it might not be a reasonable expectation. Working at that angle might be luck. 
When you're rotating with the cursor, you can snap to 45 degrees and 90 degrees by moving the cursor onto one of the lines in the rotation widget.
Overlap:
[image]
Join distance is the maximum size of gap between parts that will be joined together. 
Still playing with the model here.

closed 2026-09-29 by prusaslicer-bot[bot]