prusa3d/PrusaSlicer #14453: Sudden crash of priting and show Thermal runnaway fault

Asanga-SPLat, 2025-04-17:
### Description of the bug

I tried to print a model in MK4S printer which generated on Prusa Slicer. In the middle of printing, the printer reboots and shows thermal runaway fault. I updated the firmware of the printer into latest and I am using latest Prusa slicer. I tried several times with generating new g codes but it is failed in the middle of printing. I attached the part and generated g code.

### Project file & How to reproduce

Use following setting on Prusa Slicer to generate.

- Printer : Original Prusa MK4S 0.4 nozzle
- Filemant : PETG
- Infill : 40%
- Supports : None
- Print Settings : 0.10mm FAST DETAIL
- First layer Nozzel : 230
- First layer Bed : 110
- Other layer Nozzel : 240
- Other layer Bed : 110
     
I attached the stl file and generated g code here.

[led_align_0.4n_0.1mm_PETG_MK4S_16m.zip](https://github.com/user-attachments/files/19788635/led_align_0.4n_0.1mm_PETG_MK4S_16m.zip)

       

### Checklist of files included above

- [x] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.9.2

### Operating system

Windows 10

### Printer model

MK4S Firmware 6.2.4

rtyr, 2025-04-17:
This is likely HW problem. Please contact our tech-support via live chat at www.prusa3d.com.

closed 2025-04-17 by rtyr

Asanga-SPLat, 2025-04-28:
Sorry to say that @rtyr this is not a hardware problem. I printed other parts without any problem. Printer is working perfectly. Literally, while printing, this part reboots the printer at specific printing point. I saw system crashing. If you have time, please print this part see how it goes. 

Thanks.

rtyr, 2025-04-29:
Thermal runaway error is triggered by the firmware based on temperature readings. This error typically happens if there is some issue with the termistor or if heater (hotend or bed) is unable to maintain target temperature for some reason. 

Looking into your gcode and your filament profile, I think the issue here is, that you have a combination of very high bed temperature (110C) and also quite high print fan speed (50%). My guess is that printer is unable to maintain 110C on the bed because of the print fan running at 50% close to the bed. You should be able to see this behavior on the screen during the print (actual temp not matching target temp before restart and thermal runaway error). 

The solution is to correct your filament profile. You can significantly reduce print fan speed, reduce bed temperature or set more layers without print fan. Your profile may work fine with bigger objects, because fan speed is lower and the bed is not constantly cooled on one spot.

Asanga-SPLat, 2025-04-29:
@rtyr I appreciate if it is thermal runaway. Then I can fix it. But it is not. Forget about thermal runaway. Printing above part reboots the printer in the middle of printing. How it happens?

rtyr, 2025-04-30:
Thermal runaway resets the printer before showing the message. 

Thermal runaway was mentioned by you before editing initial issue and its very likely the cause.