prusa3d/PrusaSlicer #14454: Cannot accept cookies, can only send to connect through invisible button

Zypp, 2025-04-17:
### Description of the bug

[image]

When trying to send to connect I cannot accept cookies (all 3 buttons completely unresponsive). I tried restarting but it happens every single time. Can click the 'confirm' button which is directly below 'accept all' to send to connect anyway

### Project file & How to reproduce

Open an stl file > slice > send to connect > unable to click any of the 3 buttons in the cookie popup > can click the 'confirm' button which is directly below 'accept all' to send to connect anyway

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.9.2

### Operating system

windows 11

### Printer model

mk4s

SpaceSketch, 2025-04-17:
Happening to me as well, Windows 11, 2.9.1, PRUSA XL.

henryhbk, 2025-04-17:
**Happens on MacOS as well**. I tried accepting the cookies in Safari which made no difference.

[image]

danielfmontero, 2025-04-17:
Thanks for the trick about the invisible button. I was able to send the print to my printer.

I have the same issue.
MacOS 15.3.2 - PrusaSlicer 2.9.2

JoshuaGranger, 2025-04-18:
To get around this, use the "Prusa Connect" tab in the slicer to make your cookies selection. After doing so the prompt will go away on the "Send to Connect" screen.

[image]

TerPuc, 2025-04-22:
Dear @Zypp, we have recieved an infomation that this should be fixed. I am not able to reproduce this issue. May I kindly ask you to check on your side and let us know or close this issue? Thank you!

closed 2025-04-23 by Zypp