prusa3d/PrusaSlicer #14455: Cookies bar overlaps "Choose a printer" window in Prusa Connect

dwbmb, 2025-04-17:
### Description of the bug

When I want to send my sliced gcode into printer via Prusa Connect queue, a cookie bar overlaps bottom buttons for I can not finish print command. 

None of those buttons on cookie bar actually works, clicking on them does nothing. 

This prevents me from using Prusa Connect completely.

[image]

### Project file & How to reproduce

1. Slice a model
2. Click on "Sent to Connect" button 
3. Watch how cookies bar overlaps buttons

### Checklist of files included above

- [ ] Project file
- [x] Screenshot

### Version of PrusaSlicer

2.9.2

### Operating system

Windows 11

### Printer model

Prusa Mini+

evancharlton, 2025-04-18:
As a workaround: you can use keyboard navigation. `Tab` can get you into the options ("Set ready and print now", "Add to queue", etc). The up/down arrow key changes which option is selected, then `Tab` will move you to the "Confirm" button, which you can press with `return`.

Annoying, and I hope it's fixed very soon.

dwbmb, 2025-04-18:
Thanks for the tip. But I can not see what option is selected under that bar. I found another workaround though -> be super fast and click on what I need before cookie bar pops up :D

evancharlton, 2025-04-18:
> Thanks for the tip. But I can not see what option is selected under that bar. I found another workaround though -> be super fast and click on what I need before cookie bar pops up :D

In https://github.com/prusa3d/PrusaSlicer/issues/14454#issuecomment-2814327896, @JoshuaGranger shared that you can go to the "Prusa Connect" tab and accept the cookies there.

Hope that helps!

TerPuc, 2025-04-22:
Closing as duplicate.

closed 2025-04-22 by TerPuc