prusa3d/PrusaSlicer #14458: Side-bar taking up 3 quarters of screen

Tom6543, 2025-04-18:
### Description of the bug

Side-bar taking up 3 quarters of screen and we can't shrink it back to 1 quarter of the screen.

### Project file & How to reproduce

Don't know how.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.9.2

### Operating system

windows 11

### Printer model

Prusa XL 5 print heads

Tom6543, 2025-04-24:
Screen photos as requested.

[image]

[image]

TerPuc, 2025-04-24:
Hello @Tom6543, thank you for provided photos. From what we can see it looks like the monitor settings are not optimized. You may try to change the scale and font size.

closed 2026-09-29 by prusaslicer-bot[bot]